DLR-RM/RAFCON #2: Release?

adamwtow, 2018-06-14:
I'm wondering if you are still intending to release RAFCON?

franzlst, 2018-06-14:
Hi @adamwtow,

we still plan to make RAFCON OpenSource. Yet, we need to clarify some internal stuff before.

Please contact us via rafcon@dlr.de for your specific use case.

Rbelder, 2018-07-17:
So finally RAFCON is now Open Source. It is available on GitHub (https://github.com/DLR-RM/RAFCON), PyPi (pip install rafcon), Read the Docs (http://rafcon.readthedocs.io/en/latest/) and has its own homepage (https://DLR-RM.github.io/RAFCON/). The homepage and the documentation are getting there finishing while the next days but everything is online and can be used.

closed 2018-07-17 by Rbelder

stoertebeker23, 2018-07-18:
Nice!

uservinu, 2018-07-20:
Great News. Congratulations Y'all:+1: :100: