DLR-RM/RAFCON #4: GTK theme does not exist

chris-rogers-anki, 2018-09-20:
File "/usr/local/lib/python2.7/site-packages/rafcon/gui/config.py", line 57, in __init__
    self.configure_gtk()
  File "/usr/local/lib/python2.7/site-packages/rafcon/gui/config.py", line 82, in configure_gtk
    raise ValueError("GTK theme does not exist")
ValueError: GTK theme does not exist

ls RAFCON/source/rafcon/gui/assets/themes:
MaD       dark      templates

I put the MaD theme in there to see if that fixed it.
gtk-chtheme works for me to change gtk themes in general.

OSX 10.13.6
Python2.7

franzlst, 2018-09-21:
You are trying to run RAFCON under OSX, sounds exciting :smile: We have no experience with this, nevertheless we would like to help you.

RAFCON ships with a custom GTK theme, which should not be exchanged. This theme is copied to your system when installing RAFCON. It should either reside at
`/usr/local/share/rafcon/gui/assets/themes/dark/gtk-2.0/gtkrc` or
`$HOME/.local/share/rafcon/gui/assets/themes/dark/gtk-2.0/gtkrc`,
depending on whether you call `pip install` with `--user` or not.

Could you please check, whether `gtkrc` is at one of the named places (probably the first one)?

ct2034, 2018-10-01:
I have the exact same behavior under ubuntu xenial.
It is no OSX issue

chris-rogers-anki, 2018-10-01:
Hello, I spent some time trying and it was the python-gtkglext that didnt
work under osx.
i did get it to work without much trouble in a virtualbox ubuntu instance,
i cloned the repo and ran it out the local repo.

franzlst, 2018-10-02:
I finally found the bug! It is fixed in our local `develop` branch and will be published with the next release. For now, you should be able to easily fix it yourself directly in the code:

Open the file `utils/resources.py` if the `rafcon` package (probably in `/usr/lib/python2.7/site-packages/rafcon`). Exchange the lines 26 and 27
```python
join('usr', 'local', 'share'),
join('usr', 'share'),
```
with
```python
join(os.sep, 'usr', 'local', 'share'),
join(os.sep, 'usr', 'share'),
```
Otherwise, the paths are interpreted as relative paths (`usr/...`) instead of absolute (`/usr/...`) and the theme resource cannot be found.

@chris-rogers-anki
I really wonder, how `python-gtkglext` is related to this issue. Maybe it is related to another issue on OSX and it is great that you have found the reason.

ct2034, 2018-10-02:
ok, thanks. That solved the gtk theme issue. Now i get:
`/usr/lib/python2.7/site-packages/rafcon`

franzlst, 2018-10-03:
@ct2034:
Have you tried to follow the [installation preparations for Ubuntu 16.04](https://github.com/DLR-RM/RAFCON#installation-requirements-for-ubuntu-1604)? There it is mentioned that you need to install `python-gtkglext1`, `python-gtksourceview2` and some more packages manually before installing RAFCON.

Please tell us if that helps or if there are still complications!

franzlst, 2018-10-03:
I moved the 2nd part of the discussion (regarding the `ImportError`s to issue #6. Please post any follow ups there!

closed 2018-10-03 by franzlst

ct2034, 2018-10-03:
yes, reading the docs helps, of cause. sorry about that

franzlst, 2018-10-03:
> How much work would it be for us to get the custom gtkgextl stuff to work under OSX?

Right now, I am not fully sure for what gtkglext stuff is needed. It might be, that it is only needed for the GraphicalEditor that is bases on OpenGL. There is a second GraphicalEditor, based on gaphas, which is actually also the default editor. So if gtkglext is only needed for the OpenGL editor, it might be sufficient to remove some import or wrap them properly in `try...except`statements. I will try to check this, soon.

If you want to give it a try yourself, a good point to start at is in the [`state_machines_editor.py`](https://github.com/DLR-RM/RAFCON/blob/0009aee7be9bf046f09f30fd7dd81510216fa2e4/source/rafcon/gui/controllers/state_machines_editor.py#L49)

chris-rogers-anki, 2018-10-03:
Thank you,  Franz! I will let you know how it goes.

Has anyone combined rafcon with pybullet.org?

SebastianGB, 2018-10-04:
> Has anyone combined rafcon with pybullet.org?

@chris-rogers-anki: 
This sounds interesting! I also opened a new issue (#7) for this.

franzlst, 2018-12-17:
> How much work would it be for us to get the custom gtkgextl stuff to work under OSX?

With RAFCON [0.13.0](https://github.com/DLR-RM/RAFCON/releases/tag/0.13.0), all dependencies to OpenGL have finally been dropped. We furthermore moved from GTK+ 2 to GTK+ 3, which reduces the number of dependencies even further.