DLR-RM/RAFCON #5: Plans to Upgrade to PyGObject and Python3?

danyeaw, 2018-09-21:
I recently merged support for Python 3 in gaphas, and I am finalizing the upgrade to PyGObject now with https://github.com/gaphor/gaphas/pull/20. Do you have any plans to upgrade RAFCON as well? The [countdown](https://pythonclock.org/) is on.

It would be great to test the upgraded version of gaphas with RAFCON before releasing a new version, and I would like to coordinate with you to make sure everything goes smoothly. I feel like I am also just a bit dangerous having gone through the upgrade now, so I would be happy to share the tips I learned as well.

franzlst, 2018-09-21:
Dear @danyeaw,
I am closely following your progress in porting gaphas to both Python 3 and PyGObject and am happy that you have come that far. I am even more happy that you want to support us in porting RAFCON.

Indeed, we have plans to port RAFCON to both Python 3 and PyGObject, probably in the same order as you did. There is no schedule, yet, but with gaphas being ported, the basis is there. My hope is that we can port RAFCON within the next months.

Any help is greatly appreciated!

amolenaar, 2018-11-22:
I just tried the develop branch, but run into some issues.

A `pip3 install --user -r requirements.txt` results in an error:

    yaml-configuration requires Python '>=2.6, !=3.*' but the running Python is 3.6.7

Also installing dependencies using `pip3 install --user .` raises errors (No module named "model").

franzlst, 2018-11-23:
Sorry for the inconvenience. We haven't released a new version of `yaml-configuration`, yet. In addition, a new version of `gtkmvc` is required. I'll tell you when these have been released.

SebastianGB, 2018-11-25:
`yaml-configuration` and `gtkmvc ` are released. Python2 is working properly in the public develop branch now:

    git checkout develop
    pip2 install --user -r requirements.txt
    python2 install --user -r requirements.txt

Don't forget to put gaphas and rafcon in your PYTHONPATH, e.g.:
    
    export PYTHONPATH=$PYTHONPATH:/your/path/to/gaphas/checkout
    export PYTHONPATH=$PYTHONPATH:/your/path/to/rafcon/checkout/source

Finally start rafcon via:
    
    cd /your/path/to/rafcon/checkout
    python2 .bin/rafcon

For python < 3.6 it works analogously.

For python >= 3.6 we have to fix python-jsonconversion first, see:
https://github.com/DLR-RM/python-jsonconversion/issues/1

franzlst, 2018-11-27:
FYI:
* The `develop` branch has been updated. The installation should now e.g. work from within a (blank) virtualenv.
* `jsonconversion` now works with Python 3.6

Rbelder, 2018-11-29:
So as long as I do not like to open an issue in gaphas for this. 

@danyeaw or @amolenaar Can you update your setup.py to version 0.8.x

The develop branch of RAFCON is already on the next release 0.13.0 and already depending on ['gaphas>=0.8'](https://github.com/DLR-RM/RAFCON/commit/3a0e56fe6f274b49e86f2ed2255898cec7d06220).

I manually changed it (and install it locally) and tested it for our local installation and build. Except of this the routine of `python setup.py install --user` was working fine.

Rbelder, 2018-11-29:
> I updated it to 1.0.0 just now, since this is definitely a breaking change. I think we will be ready to release a new version to pypi once we resolve a segmentation fault issue that we are getting from Gtk+ on our Travis CI tests.

OK that is good, too. The issue with the seg-fault I have seen, too, good luck with this. 

Our tests succeed but we still discover some new issues that delay the first final gtk3 and python3 based RAFCON release on pypi. Most likely before the release on pypi we will do an internal release, to avoid major problems, and upload it with a delay. Hopefully we make it this year.

danyeaw, 2018-12-05:
gaphas 1.0.0rc1 is ready for testing on PyPi using `pip install gaphas --pre`. Please let me know if anyone sees any issues.

franzlst, 2018-12-05:
Yeah! I adapted our requirements and now everything seems to work smoothly :smile: If you want, you can check it with the latest commit on the `develop` branch.

franzlst, 2018-12-17:
RAFCON 0.13.0 has [just been released](https://github.com/DLR-RM/RAFCON/releases/tag/0.13.0)!

Now you can use RAFCON with both Python 2.3 and Python >3.4. Furthermore, the GUI is based on PyGObject!

closed 2018-12-17 by franzlst