DLR-RM/RAFCON #7: Experiences with pybullet

SebastianGB, 2018-10-04:
_Originally posted by @chris-rogers-anki in https://github.com/DLR-RM/RAFCON/issues/4#issuecomment-426729853_

...
Has anyone combined rafcon with pybullet.org?

SebastianGB, 2018-10-04:
@chris-rogers-anki: 
This sounds interesting! In our institute we use bullet also for robot simulation, most of the time in the Gazebo simulator. But, as far as I know, we didn't use pybullet (yet).

Can you explain your use case a little bit?

chris-rogers-anki, 2018-10-04:
Hello,
We (Anki.com) make a consumer robot called Vector (launches next Friday)
I'm on the art tools team.  We program animation tools for our robot.  We
have a state machine editor we built, but its very simple compared to yours.
We used Webots to do the robot sims but are looking at other possibilities,
and it seems like pybullet (bullet with python bindings and a focus on
robotic sim and reinforcement learning) and rafcon are worth looking into.

SebastianGB, 2018-10-08:
Sounds great! If there are any bugs to fix or features you need, feel free to create a another issue!
If there is any more information about how you use pybullet with rafcon, feel free to post this here as well!

commit f86c0c55 (2022-09-06): Merge branch 'feat/create-data-flow-automatically' into 'develop'

Create the data flows & data ports automatically in the nested states

Closes #7

See merge request dev/sys/tci/rafcon!72
closed 2019-06-04 by SebastianGB