DLR-RM/RAFCON #9: Run with hidden GUI cause "maximum recursion depth exceeded" error

Rbelder, 2019-01-08:
The issue could also be situated in the [gaphas repo](https://github.com/gaphor/gaphas) but I wanted to keep track here.

The issue occurs in a special run configuration of our tests. The tests as well as the RAFCON GUI can be run in a minimized GUI mode to e.g. hide tests runs or simply start minimized RAFCON GUI instances. The example commands to reproduce is as follows (see the traceback below):

```
cd /RAFCON_REPO/tests
RAFCON_LIBRARY_PATH= RAFCON_START_MINIMIZED=1 py.test -s gui
```

It would be good if one of the core developer of gaphas @danyeaw or @amolenaar could look at it, too.

It is a corner case but maybe it becomes at some point a bigger problem. With gtk2 and respective gaphas version we have not had this issue.

```
log.py                     148 ERROR    Unexpected error
Traceback (most recent call last):
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/utils/log.py", line 143, in wrapper
    return function(*args, **kwargs)
  File "/home_local/in_progress/rafcon/repo/rafcon/tests/gui/widget/test_menu_bar.py", line 329, in trigger_gui_signals
    call_gui_callback(menubar_ctrl.on_refresh_all_activate, None, None, True)
  File "/home_local/in_progress/rafcon/repo/rafcon/tests/testing_utils.py", line 109, in call_gui_callback
    return rafcon.utils.gui_functions.call_gui_callback(*((callback, ) + args), **kwargs)
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/utils/gui_functions.py", line 72, in call_gui_callback
    raise_(e_type, e_value, e_traceback)
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/utils/gui_functions.py", line 49, in fun
    result = callback(*args)
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/controllers/menu_bar.py", line 423, in on_refresh_all_activate
    gui_helper_state_machine.refresh_all(force=force)
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/helpers/state_machine.py", line 598, in refresh_all
    state_machines_editor_ctrl.refresh_all_state_machines()
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/controllers/state_machines_editor.py", line 539, in refresh_all_state_machines
    self.refresh_state_machines(list(self.model.state_machine_manager.state_machines.keys()))
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/controllers/state_machines_editor.py", line 506, in refresh_state_machines
    was_closed = self.on_close_clicked(None, self.model.state_machines[sm_id], None, force=True)
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/controllers/state_machines_editor.py", line 412, in on_close_clicked
    remove_state_machine_m()
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/controllers/state_machines_editor.py", line 367, in remove_state_machine_m
    self.model.state_machine_manager.remove_state_machine(state_machine_id)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gtkmvc3/gtkmvco/gtkmvc3/observable.py", line 48, in wrapper
    self._notify_method_after(self, _func.__name__, res, args, kwargs)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gtkmvc3/gtkmvco/gtkmvc3/support/wrappers.py", line 84, in _notify_method_after
    args, kwargs)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gtkmvc3/gtkmvco/gtkmvc3/model.py", line 754, in notify_method_after_change
    self, prop_name, info)
  File "/home_local/in_progress/rafcon/repo/rafcon/tests/testing_utils.py", line 471, in __patched__notify_observer__
    return Model.__notify_observer__(self, observer, method, *args, **kwargs)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gtkmvc3/gtkmvco/gtkmvc3/model.py", line 601, in __notify_observer__
    return method(*args, **kwargs)
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/models/state_machine_manager.py", line 108, in model_changed
    sm_m.prepare_destruction()
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/models/state_machine.py", line 154, in prepare_destruction
    self.destruction_signal.emit()
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gtkmvc3/gtkmvco/gtkmvc3/observable.py", line 97, in emit
    model.notify_signal_emit(name, arg)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gtkmvc3/gtkmvco/gtkmvc3/model.py", line 784, in notify_signal_emit
    self, prop_name, info)
  File "/home_local/in_progress/rafcon/repo/rafcon/tests/testing_utils.py", line 457, in __patched__notify_observer__
    return Model.__notify_observer__(self, observer, method, *args, **kwargs)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gtkmvc3/gtkmvco/gtkmvc3/model.py", line 601, in __notify_observer__
    return method(*args, **kwargs)
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/controllers/graphical_editor_gaphas.py", line 322, in state_machine_destruction
    self.canvas.get_view_for_model(self.root_state_m).remove()
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/mygaphas/items/state.py", line 193, in remove
    child.remove()
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/mygaphas/items/state.py", line 193, in remove
    child.remove()
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/mygaphas/items/state.py", line 193, in remove
    child.remove()
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/mygaphas/items/state.py", line 193, in remove
    child.remove()
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/mygaphas/items/state.py", line 923, in remove
    self.canvas.remove(self)
  File "/home_local/in_progress/rafcon/repo/rafcon/source/rafcon/gui/mygaphas/canvas.py", line 68, in remove
    super(MyCanvas, self).remove(item)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/canvas.py", line 156, in remove
    self._remove(item)
  File "<decorator-gen-12>", line 2, in _remove
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/state.py", line 77, in wrapper
    return func(*args, **kwargs)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/canvas.py", line 136, in _remove
    self._update_views(removed_items=(item,))
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/canvas.py", line 847, in _update_views
    v.request_update(dirty_items, dirty_matrix_items, removed_items)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/view.py", line 739, in request_update
    self._qtree.remove(item)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/quadtree.py", line 196, in remove
    self._bucket.find_bucket(clipped_bounds).remove(item)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/quadtree.py", line 352, in find_bucket
    return self._buckets[index].find_bucket(bounds)
  
.... 

  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/quadtree.py", line 352, in find_bucket
    return self._buckets[index].find_bucket(bounds)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/quadtree.py", line 352, in find_bucket
    return self._buckets[index].find_bucket(bounds)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/quadtree.py", line 352, in find_bucket
    return self._buckets[index].find_bucket(bounds)
  File "/home_local/in_progress/rafcon/foreign_packages/repos/gaphas/gaphas/quadtree.py", line 343, in find_bucket
    if x >= cx:
RuntimeError: maximum recursion depth exceeded in cmp
```

Rbelder, 2019-01-08:
While the test run it can be observed that the recursion is there several times and at some point the tests stops to continues.

Note: If the flag `RAFCON_START_MINIMIZED=1` is not set the tests can be run without causing this issue.

franzlst, 2019-01-09:
I cannot reproduce this. Sounds like a timing/race condition error. If the GUI is not shown, some code might be executed faster. In my case, I get a lot of messages "The destruct of gaphas items has to be fixed!" (both with and without GUI hidden).

SebastianGB, 2019-06-04:
I cannot reproduce this either. Please reopen, with some exact steps to reproduce.

closed 2019-06-04 by SebastianGB