DLR-RM/RAFCON #10: Cryptic error message during state machine loading

SebastianGB, 2019-02-12:
When opening a state machine in which an empty folder resides or leftovers of a merge conflict are located the following error message occurs during loading.

```ERROR - core.storage.storage: Error while loading state data: Expecting property name: line 3 column 1 (char 47)
Traceback (most recent call last):
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/core/storage/storage.py", line 397, in load_state_recursively
    state_info = load_data_file(path_core_data)
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/core/storage/storage.py", line 478, in load_data_file
    return storage_utils.load_objects_from_json(path_of_file)
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/utils/storage_utils.py", line 126, in load_objects_from_json
    result = json.load(f, cls=JSONObjectDecoder, substitute_modules=substitute_modules)
  File "/usr/lib64/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib64/python2.7/json/__init__.py", line 352, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "/usr/lib64/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib64/python2.7/json/decoder.py", line 380, in raw_decode
    obj, end = self.scan_once(s, idx)
  File "/volume/software/common/packages/python-jsonconversion/0.2.9/src/jsonconversion/decoder.py", line 64, in _dict_to_qualified_object
    return cls.from_dict(dictionary)
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/core/states/library_state.py", line 375, in from_dict
    output_data_port_runtime_values, use_runtime_value_output_data_ports)
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/core/states/library_state.py", line 98, in __init__
    lib_version, state_copy = library_manager.get_library_state_copy_instance(self.lib_os_path)
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/core/library_manager.py", line 397, in get_library_state_copy_instance
    state_machine = storage.load_state_machine_from_path(lib_os_path)
  File "/volume/software/common/packages/rafcon_execution_hooks_plugin/0.2.0/lib/python2.7/execution_hooks/hooks.py", line 45, in load_state_machine_from_path_with_hooks
    state_machine = old_load_state_machine_from_path(*args, **kwargs)
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/utils/timer.py", line 13, in func_wrapper
    return_value = func(*args, **kwargs)
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/core/storage/storage.py", line 301, in load_state_machine_from_path
    state_machine_dict = storage_utils.load_objects_from_json(state_machine_file_path)
  File "/volume/software/common/packages/rafcon/0.13.5/source/rafcon/utils/storage_utils.py", line 126, in load_objects_from_json
    result = json.load(f, cls=JSONObjectDecoder, substitute_modules=substitute_modules)
  File "/usr/lib64/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib64/python2.7/json/__init__.py", line 352, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "/usr/lib64/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib64/python2.7/json/decoder.py", line 380, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting property name: line 3 column 1 (char 47)
10:37:44 ERROR - gui.helpers.state_machine: Error while trying to open state machine: 'NoneType' object has no attribute 'name'
10:37:44 ERROR - gui.controllers.library_tree: Could not open library: 'NoneType' object has no attribute 'file_system_path'
```
We should provide a nicer error message in order to help the user to find the problem.

franzlst, 2019-05-07:
I think this is improved by 96343f0d07f8ba3d48d46a96b9729241b339528a, 647fdd8229bc94c9cc8c7cf655496d0de258272a and fb9602013c6cb842b7d384bdd01c0da1ade62988

There should be no more crash and only a single error message. Yet, the error message will probably still not point to the source of the problem.

sebastian-brunner, 2020-07-29:
Yes there is only one single error message now:

```
Traceback (most recent call last):
  File "manage.py", line 22, in <module>
    execute_from_command_line(sys.argv)
  File "/home/sb/code/ar-gui/web_gui/backend/venv/lib/python2.7/site-packages/django/core/management/__init__.py", line 364, in execute_from_command_line
    utility.execute()
  File "/home/sb/code/ar-gui/web_gui/backend/venv/lib/python2.7/site-packages/django/core/management/__init__.py", line 338, in execute
    django.setup()
  File "/home/sb/code/ar-gui/web_gui/backend/venv/lib/python2.7/site-packages/django/__init__.py", line 27, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/sb/code/ar-gui/web_gui/backend/venv/lib/python2.7/site-packages/django/apps/registry.py", line 116, in populate
    app_config.ready()
  File "/home/sb/code/ar-gui/web_gui/backend/art_app/apps.py", line 22, in ready
    art_app.core.core_manager.core_manager_singleton = CoreManager()
  File "/home/sb/code/ar-gui/web_gui/backend/art_app/core/core_manager.py", line 43, in __init__
    self._skill_manager = SkillManager(self.rafcon_initialized)
  File "/home/sb/code/ar-gui/web_gui/backend/art_app/core/skill_manager.py", line 42, in __init__
    library_state = library_manager.get_library_instance(skill_root, library_key)
  File "/home/sb/.local/lib/python2.7/site-packages/rafcon/core/library_manager.py", line 371, in get_library_instance
    return LibraryState(library_path, library_name, "0.1")
  File "/home/sb/.local/lib/python2.7/site-packages/rafcon/core/states/library_state.py", line 103, in __init__
    lib_version, state_copy = library_manager.get_library_state_copy_instance(self.lib_os_path)
  File "/home/sb/.local/lib/python2.7/site-packages/rafcon/core/library_manager.py", line 387, in get_library_state_copy_instance
    if lib_os_path in self._loaded_libraries:
TypeError: unhashable type: 'OrderedDict'
```

As you said: it is still misleading.

sebastian-brunner, 2020-07-29:
credits go to @stefanbuettner