DLR-RM/RAFCON #12: Error when switching from python2 to python3

SebastianGB, 2019-03-09:
Stacktrace:
```
2019-03-09 11:03:09:    ERROR - rafcon.gui.runtime_config:  Could not read from config /home/basti/.config/rafcon/runtime_config.yaml, using temporary default configuration. Error: while constructing a mapping
  in "/home/basti/.config/rafcon/runtime_config.yaml", line 18, column 13
found unhashable key
  in "/home/basti/.config/rafcon/runtime_config.yaml", line 18, column 15
```

The specified line contains: `!!python/object/new:future.types.newstr.newstr`
So there is something screwed up with string types from future.

Steps to reproduce:
- open rafcon with python 2.7
- open a state machine
- close
- open rafcon with python 3.x
- error occurs directly when restoring session

franzlst, 2019-03-11:
I have also observed a broken `runtime_config.yaml` quite often, but was never able to reproduce the steps to break it. So I'm very happy that you have finally found that out! Now we will hopefully be able to fix this soon.

franzlst, 2019-05-07:
Fixed in da56071

closed 2019-05-07 by franzlst