DLR-RM/RAFCON #14: Session restore test cannot be run in parallel

franzlst, 2019-03-19:
The `test_restore_session` test sometimes produces error when run in parallel to other tests.

Here is a build (on our internal Jenkins) that fails due to this: https://rmc-jenkins.robotic.dlr.de/jenkins/job/common/job/rafcon/view/change-requests/job/PR-707-merge/4/consoleFull

Error output:
```
09:24:47 tests/gui/widget/test_restore_session.py::test_restore_session FAILED    [ 92%]
09:24:47 
09:24:47 =================================== FAILURES ===================================
09:24:47 _____________________________ test_restore_session _____________________________
09:24:47 
09:24:47 caplog = <_pytest.logging.LogCaptureFixture object at 0x7fd71d31f790>
09:24:47 
09:24:47     @pytest.mark.timeout(20)
09:24:47     def test_restore_session(caplog):
09:24:47         from rafcon.core.storage import storage
09:24:47     
09:24:47         change_in_gui_config = {'AUTO_BACKUP_ENABLED': False, 'HISTORY_ENABLED': False,
09:24:47                                 'SESSION_RESTORE_ENABLED': True}
09:24:47     
09:24:47         # first run
09:24:47         libraries = {"ros": join(testing_utils.EXAMPLES_PATH, "libraries", "ros_libraries"),
09:24:47                      "turtle_libraries": join(testing_utils.EXAMPLES_PATH, "libraries", "turtle_libraries"),
09:24:47                      "generic": join(testing_utils.LIBRARY_SM_PATH, "generic")}
09:24:47         testing_utils.run_gui(gui_config=change_in_gui_config, libraries=libraries)
09:24:47         try:
09:24:47             open_state_machines = {'list_of_hash_path_tab_page_number_tuple': [], 'selected_sm_page_number': None}
09:24:47             trigger_gui_signals_first_run(open_state_machines)
09:24:47         finally:
09:24:47             testing_utils.close_gui(force_quit=False)
09:24:47             testing_utils.shutdown_environment(caplog=caplog, expected_warnings=0, expected_errors=0)
09:24:47     
09:24:47         # second run
09:24:47         libraries = {"ros": join(testing_utils.EXAMPLES_PATH, "libraries", "ros_libraries"),
09:24:47                      "turtle_libraries": join(testing_utils.EXAMPLES_PATH, "libraries", "turtle_libraries"),
09:24:47                      "generic": join(testing_utils.LIBRARY_SM_PATH, "generic")}
09:24:47         testing_utils.run_gui(gui_config=change_in_gui_config, libraries=libraries)
09:24:47     
09:24:47         try:
09:24:47             final_open_state_machines = {'list_of_hash_path_tab_page_number_tuple': [], 'selected_sm_page_number': None}
09:24:47             trigger_gui_signals_second_run(final_open_state_machines)
09:24:47         except:
09:24:47             raise
09:24:47         finally:
09:24:47             testing_utils.close_gui()
09:24:47     
09:24:47         print(open_state_machines)
09:24:47         print(final_open_state_machines)
09:24:47     
09:24:47         # test selection, page number and path
09:24:47         # TODO find if there is a proper hash value test
09:24:47         # TODO find out why core and gui hashes are changing !!! not even fully deterministic !!!
09:24:47         # TODO find out why dirty flag is once wrong when AUTO_BACKUP is enabled in parallel
09:24:47         #      (is connected to direct storing while opening)
09:24:47         assert open_state_machines['selected_sm_page_number'] == final_open_state_machines['selected_sm_page_number']
09:24:47     
09:24:47         final_tuple_list = final_open_state_machines['list_of_hash_path_tab_page_number_tuple']
09:24:47     
09:24:47         assert open_state_machines['selection_state_machine'] == final_open_state_machines['selection_state_machine']
09:24:47     
09:24:47         order_of_pages_to_be_dirty = [False, True, False, False, True, False, True]
09:24:47         for index, sm_tuple in enumerate(open_state_machines['list_of_hash_path_tab_page_number_tuple']):
09:24:47             assert index == sm_tuple[PAGE_NUMBER_INDEX]
09:24:47             if not final_tuple_list[index][CORE_HASH_INDEX] == sm_tuple[CORE_HASH_INDEX]:
09:24:47                 print("CORE hashes page {4} are not equal: {0} != {1}, path: {2} {3}" \
09:24:47                       "".format(final_tuple_list[index][CORE_HASH_INDEX], sm_tuple[CORE_HASH_INDEX],
09:24:47                                 sm_tuple[PATH_INDEX], sm_tuple[MARKED_DIRTY_INDEX], sm_tuple[PAGE_NUMBER_INDEX]))
09:24:47                 if sm_tuple[PATH_INDEX]:
09:24:47                     sm_file_path = join(sm_tuple[PATH_INDEX], storage.STATEMACHINE_FILE)
09:24:47                     if exists(sm_tuple[PATH_INDEX]) and exists(sm_file_path):
09:24:47                         print("sm_file_path: ", sm_file_path)
09:24:47                         print(storage.load_data_file(join(sm_tuple[PATH_INDEX], storage.STATEMACHINE_FILE)))
09:24:47                     else:
09:24:47                         print("does not exist sm_file_path ", sm_file_path)
09:24:47                 else:
09:24:47                     print("state machine was NOT stored")
09:24:47             assert final_tuple_list[index][CORE_HASH_INDEX] == sm_tuple[CORE_HASH_INDEX]
09:24:47 >           assert final_tuple_list[index][PATH_INDEX] == sm_tuple[PATH_INDEX]
09:24:47 E           AssertionError: assert '/home_local/...urtle_demo_sm' == '/home_local/j...urtle_demo_sm'
09:24:47 E             - /home_local/jenkins/common_rafcon_PR-707-head/tests/../share/examples/tutorials/basic_turtle_demo_sm
09:24:47 E             ?                                          ^ ^^
09:24:47 E             + /home_local/jenkins/common_rafcon_PR-707-merge/tests/../share/examples/tutorials/basic_turtle_demo_sm
09:24:47 E             ?                                          ^ ^^^
09:24:47 
09:24:47 tests/gui/widget/test_restore_session.py:326: AssertionError
09:24:47 ----------------------------- Captured stdout call -----------------------------
```

For this build, the `pr-head` and `pr-merge` build were running in parallel.

Maybe this can be fixed by fixing #13.

franzlst, 2019-03-19:
I fixed #13 (in the internal jenkins branch), yet this issue remains, though with a different error:

https://rmc-jenkins.robotic.dlr.de/jenkins/job/common/job/rafcon/view/change-requests/job/PR-707-head/19/console

```

Rbelder, 2019-03-19:
I have to look into it. 

But #13 shows some behavior that reminds me to some issues I have seen concerning gui_config.yaml and runtime_config.yaml and not clear defined paths for those.
And this second traceback here is more or less what I would assume to happen. If I am right this is maybe not directly a `session restore` test issue but an issue of the configuration_yaml package or its usage by RAFCON.

I will look into it and lets have a direct chat about it tomorrow.

franzlst, 2019-03-20:
Sorry, the new error came from the `resave_state_machines.py` script and has nothing to do with session restore. I will try to fix it.

As it looks to me now, with fixing #13, also this issue is fixed.

franzlst, 2019-03-20:
I close this issue, as it hasn't occurred since fixing #13.

closed 2019-03-20 by franzlst