DLR-RM/RAFCON #17: Basic turtle demo and ros

pbsgoncalves, 2019-04-11:
trying to run the demo, i have experienced the following two items related to python, I believe:

1.
changed the init_ros_node script line 12, from
except Exception , e:
to
except Exception as e:

2.
the following error appears:
ERROR - core.states.execution_state: ExecutionState with name 'init_ros_node' and id 'INIT_ROS_NODE' had an internal error: ImportError: dynamic module does not define module export function (PyInit__tf2)
Traceback (most recent call last):
  File "script.py (20)", line 2, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)

can you help!

franzlst, 2019-04-11:
Hi,

regarding your first point: `except Exception , e:` is valid in Python <= 2.7. This leads me to the question: are you maybe using Python 3?

Because, reagarding your 2nd point, ROS does not support Python 3. In any case, the stacktrace you posted stems from ROS, not RAFCON. So I am quite sure that this has nothing to to with RAFCON.

Googleing for the error message indicates that my assumption is probably right and you are trying to use ROS with Python 3, see for example [this issue](https://github.com/whoenig/crazyflie_ros/issues/82) or [this issue](https://github.com/ros/geometry2/issues/259).

I hope this helps!

SebastianGB, 2019-04-12:
Please use python2 for ROS. Python3 is not officially supported, even not for [melodic](http://www.ros.org/reps/rep-0003.html#melodic-morenia-may-2018-may-2023).

Also see [this link](https://answers.ros.org/question/237613/how-to-define-ros-kinetic-to-use-python3-instead-of-python27/) for more information.

pbsgoncalves, 2019-05-02:
Hi,
I am using Python 2.7.12
the output of command python --version

This is quite a strange behaviour because, in a terminal, I open a python console an I can import the library, using the command:
import tf

also from the command line:
python -c "import tf"
python2.7 -c "import tf"

only when I import the library (import tf)  in the RAFCON scripts the error occurs.
also tried to import the library in the example 99 bottles of beer and the same error occurs.

do RAFCON needs any special environment?
what do you mean by these lines
----------------
rmpm_do env ros.indigo.desktop > /tmp/desktop.env
source /tmp/desktop.env
rmpm_do env rafcon > /tmp/rafcon.env
source /tmp/rafcon.env
--------------------
in:
https://rafcon.readthedocs.io/en/latest/tutorials.html#starting-the-basic-turtle-demo-state-machine-using-ros

thanks in advance.

Rbelder, 2019-05-02:
Have you resolved your issue @pbsgoncalves? we have recognized your last comment which vanished very fast. 

We assume you found out that the documentation was about an DLR internal environment. 

If you still have the issue give some more insides and check if you have confirmed the mentioned solutions of @SebastianGB and @franzlst to be in place.

SebastianGB, 2019-05-07:
If you have any further questions, feel free to reopen this issue.

closed 2019-05-07 by SebastianGB