DLR-RM/RAFCON #18: State machines with library states cannot be opened if show flag is set to True

SebastianGB, 2019-04-25:
Steps to reproduce:

- create state machine
- add arbitrary library state and set start transition
- set show flag to true
- save state machine
- close state machine
- try to reopen

The following stacktrace occurs:

```
Traceback (most recent call last):
  File "/home/suer_ch/USERSTORE/repositories/gtkmvc3/gtkmvco/gtkmvc3/controller.py", line 176, in _idle_register_view
    self.register_view(view)
  File "/home/suer_ch/USERSTORE/repositories/rafcon/source/rafcon/gui/controllers/graphical_editor_gaphas.py", line 109, in register_view
    self.setup_canvas()
  File "/home/suer_ch/USERSTORE/repositories/rafcon/source/rafcon/gui/controllers/graphical_editor_gaphas.py", line 731, in setup_canvas
    self.add_state_view_for_model(self.root_state_m, rel_pos=gui_helper_meta_data.STATE_DEFAULT_REL_POS)
  File "/home/suer_ch/USERSTORE/repositories/rafcon/source/rafcon/core/decorators.py", line 53, in func_wrapper
    return_value = func(*args, **kwargs)
  File "/home/suer_ch/USERSTORE/repositories/rafcon/source/rafcon/gui/controllers/graphical_editor_gaphas.py", line 823, in add_state_view_for_model
    self.add_state_view_for_model(child_state_m, state_v, child_rel_pos, child_size, hierarchy_level + 1)
  File "/home/suer_ch/USERSTORE/repositories/rafcon/source/rafcon/core/decorators.py", line 53, in func_wrapper
    return_value = func(*args, **kwargs)
  File "/home/suer_ch/USERSTORE/repositories/rafcon/source/rafcon/gui/controllers/graphical_editor_gaphas.py", line 808, in add_state_view_for_model
    gui_helper_meta_data.scale_library_content(state_m)
  File "/home/suer_ch/USERSTORE/repositories/rafcon/source/rafcon/gui/helpers/meta_data.py", line 447, in scale_library_content
    models_dict[state_element_key] = {elem.core_element.core_element_id: elem for elem in state_element_list}
```

franzlst, 2019-04-26:
The error already occurs when enabling the show flag. It is already fixed in an internal branch and will be published soon.

Rbelder, 2019-04-26:
Small note: The Bug was created after version 0.13.7 was released. So the bug never was in the master and only people that use the develop branch were effected.

Rbelder, 2019-04-26:
Fix is merged into develop branch. The issue is closed. As already said if you used always the released version you never have had to face this issue.

closed 2019-04-26 by Rbelder