DLR-RM/RAFCON #19: Annoying warnings visualized in RAFCON console

SebastianGB, 2019-05-07:
Warnings like:

```
2019-05-07 10:20:25:  WARNING - py.warnings:  /home/brun_sb/.local/lib/python2.7/site-packages/gi/overrides/Gtk.py:1629: Warning: g_object_notify: assertion 'G_IS_OBJECT (object)' failed
  return _Gtk_main(*args, **kwargs)
```

are now also shown in the rafcon console, which is quite annoying. 

This is probably related to the warnings module, we introduced in March.

And they are even shown several times. Shouldn't they be only shown once?

franzlst, 2019-06-03:
I get some `DeprecationWarning`s when running RAFCON, but not the warning you have specified. When does this message occur?

Do you have the `PYTHONWARNINGS` env var set?

franzlst, 2019-06-04:
I can now reproduce this. The three warnings appear, when a state type change is performed.

SebastianGB, 2019-06-04:
Yes, it only happens there.

JohannesErnst, 2024-12-13:
Cannot reproduce, might be fixed.

closed 2024-12-13 by JohannesErnst