DLR-RM/RAFCON #26: python3's "__pychace__" folder chrashes loading of examples (maybe more)

Kaiser1401, 2019-08-27:
Hi,

I just installed rafcon in a python 3 environment (`pip3 install rafcon --user`) on a system it has not been running before. Executing `~/.local/bin/rafcon` starts the GUI. When opening anything from the library section (examples, libs, etc) I get the following error:

>  `core.storage.storage: Error while loading state data: Data file not found: /home/USER/.local/share/rafcon/examples/libraries/turtle_libraries/clear_field/clear_field_CLEAR_FIELD/__pycache__/meta.json
Traceback (most recent call last):
  File "/home/USER/.local/lib/python3.6/site-packages/rafcon/core/storage/storage.py", line 401, in load_state_recursively
    state_info = load_data_file(path_core_data)
  File "/home/USER/.local/lib/python3.6/site-packages/rafcon/core/storage/storage.py", line 495, in load_data_file
    raise ValueError("Data file not found: {0}".format(path_of_file))
ValueError: Data file not found: /home/USER/.local/share/rafcon/examples/libraries/turtle_libraries/clear_field/clear_field_CLEAR_FIELD/__pycache__/meta.json`

Seeing these files are not there, I started looking in the mentioned python files, and it seems it's trying to load (parts of) state machines recursively through all folders. Python3 manages to put a cache folder next to python files though (as opposed to python2), which of course has no valid things to load, but the code still tries, and fails.

After cleaning the cache files (running `py3clean ~/.local/share/rafcon/`) and preventing python from creating caches (insert `sys.dont_write_bytecode = True` in `~/.local/bin/rafcon`) the examples open just fine.

I feel preventing the caches is the wrong way though. Maybe there is another solution?

Note: Reproducing this issue, I noticed, that when I remove the added line in `~/.local/bin/rafcon` the problem does not reoccur before I reinstall rafcon.

(btw: This is ubuntu18.04 running in win10's WSL, wish there was an easy way to set it up running it under windows (that i can also explain to non-developers), but had no luck with gtk / cairo so far)

sebastian-brunner, 2019-08-28:
Hi @Kaiser1401,
thanks a lot for the issue and the steps to reproduce! I think I can remember that we had something similar already.
Thank you also for your temporary fix! It's nice that it works. However, as you mention yourself, this might not be most beautiful solution.

One problem is, that I cannot reproduce this on my machines. Right now I test on a Ubuntu18.04 with both python 3.7 and 3.6, but I cannot reproduce this. I also used to work on Open Suse Leap, where I don't have the problem either. I am using the latest rafcon version: 14.0

Could you tell me which python version you are using? And which rafcon version you installed?
Furthermore, it would be great if you could check if these `__pycache__` folders are already created during installation of rafcon or during the first launch of rafcon.

Concerning windows: We already had a few requests for supporting Windows as well (which would be possible as we fully rely on gtk which is cross platform). If you seriously plan to use rafcon for Windows, we might convince the resource lead at our institute to setup Windows support. Are you allowed to tell what you would use RAFCON for? And how many people would use it approx.? This might support our request.

franzlst, 2019-08-28:
@sebastian-brunner: Maybe we cannot reproduce this. However, fixing this should be relatively easy nevertheless. Either we skip folders called `__pycache__`, or (what I would vote for), skip all folders not containing a `core_data.json`.

I'm on parental leave, so I don't have time to fix this.

Kaiser1401, 2019-08-28:
* Python version is 3.6.8
* rafcon should be 0.14 (according to pip installing `rafcon-0.14.0-py2.py3-none-any.whl`  )
* cache folders are already created on install

I read, that the cache folder location can be changed [from python 3.8 on](https://docs.python.org/3.8/whatsnew/3.8.html) but that might not be too helpful for all the in-between versions. I agree that loading just verified folders is probably the better solution to ignoring `__pycache__` folders.

For now windows use would more be convenience thing to allow colleagues an easy / introduction as many of them run windows. Especially an easy setup might encourage them to give it a chance. I can't estimate a user count as i have to look into it more to estimate the usability in our robotics and research projects (Is there any ready solution to interface _universal robots_' devices by chance?).

(also, greetings to Oberpfaffenhofen from Trondheim :p)

closed 2019-08-28 by sebastian-brunner

sebastian-brunner, 2019-08-28:
Thanks for your information @Kaiser1401.

I was able to reproduce the error. I had just set the RAFCON_LIB_PATH env var to my local development folder, which did not contain the `__pycache__` folders.

I just released a new version. As @franzlst  suggested, I now skip all folders that do not contain a valid state core file (`core_data.json`).

Concerning windows. Yes you are right, if we would support Windows we could get more popularity. We will discuss this in our institute.
Concerning universal robots ... al least we don't have a ready solution. We build our own robots :P (see https://www.dlr.de/rm/en/desktopdefault.aspx/tabid-11370/).
But it would be great that (if you build one) upload it to github, so that other people can profit as well.

Greetings back to Trondheim :-). Do you know our institute?

sebastian-brunner, 2019-09-09:
Does the new version solve your issue @Kaiser1401 ?

Kaiser1401, 2019-10-02:
👍 Libraries can now be accessed after installation without the workaround

franzlst, 2019-10-07:
Thank you for your feedback! :+1:

commit 71e3bb2f (2019-08-28): fix(storage): resolves #26

python3's "__pycache__" folder chrashes loading of examples

fix: all folders not containing a core_data.json are skipped upon
loading a state machine