DLR-RM/RAFCON #27: Align on Dropping Support for Python2

danyeaw, 2019-10-05:
[Gaphas](https://github.com/gaphor/gaphas) would like to use Python 3 to its full potential, and we don’t intend to maintain Python 2 compatibility indefinitely. CPython is dropping support for legacy Python on January 1, 2020. I would like to have Gaphas drop support for Python 2 before that date if possible.

Are there any plans to drop Python 2 support in RAFCON?

franzlst, 2019-10-07:
Hi @danyeaw,

currently, we have no plans to drop the support for Python 2 (although it would probably make sense). Nevertheless, I see no reason why Gaphas shouldn't drop the support (in a new minor version). RAFCON species the version number of Gaphas it requires and could even do so depending on the Python interpreter. So there should be no conflicts or issues!

danyeaw, 2019-10-08:
Ok great, thanks!

closed 2019-10-08 by danyeaw

sebastian-brunner, 2019-10-09:
Thanks for your information! Personally, I would like to move to pure Python 3 support as well. However, we are using several middlewares (in combination with RAFCON), which purely rely on Python 2 - most notably ROS indigo/melodic and 'links and nodes' (see http://blog.pal-robotics.com/wp-content/uploads/2014/09/Schmidt_WSAH.pdf).
As @franzlst said, it would be totally fine if you drop the Python 2 support in your next minor (or major).