DLR-RM/RAFCON #30: DEBUG CONSOLE panel not showing any output

LJMP, 2020-01-14:
- system is Ubuntu 16 LTS mini install with xfce desktop minimal (Linux 4.4.0-171-generic #200-Ubuntu SMP x86_64 GNU/Linux)
- rafcon 0.14.1 (installed from pip)

when I start rafcon the DEBUG CONSOLE panel shows
```
DEBUG CONSOLE

INFO - gui.controllers.main_window: Ready
```

when I run some state machine, no new logger prints / python errors / anything get added to the console
when I click any of the :ballot_box_with_check: checkboxes (verbose, debug, info, ...) the one and only line _INFO - gui.controllers.main_window: Ready_ also disappears and the output stays just blank with a cursor.

The shell I run the _rafcon_ command in shows logger print just fine :woman_shrugging: 
but I noticed errors like:
`(rafcon:4164): Gtk-WARNING **: Allocating size to main_window 0x37324d0 without calling gtk_widget_get_preferred_width/height(). How does the code know the size to allocate?`
`(rafcon:3834): Pango-CRITICAL **: pango <whatever assertion> != NULL' failed`

**How can I fix logger prints / errors getting displayed in DEBUG CONSOLE?**

sebastian-brunner, 2020-01-15:
Hi @LJMP,
which python version are you using?

sebastian-brunner, 2020-01-15:
Unfortunately, I cannot reproduce the issue on my setup (xfce on my Ubuntu 16.04 VM).

Did you install rafcon as described in our README? https://github.com/DLR-RM/RAFCON

You need gtk > 3.20 for rafcon > 0.12.x . You can compile gtk yourself, however, this is quite cumbersome. Thus, we just described the approach to install gnome3 in the README, which ships with the required gtk version.

sebastian-brunner, 2020-01-15:
I just checked: both python2.7 + python3.5 works fine in Ubuntu 16.04 + xfce.

Only the default source editor style looks weird. You should change it (SOURCE_EDITOR_STYLE in config) using the instructions given in:
https://rafcon.readthedocs.io/en/latest/configuration.html

LJMP, 2020-01-15:
`python -V` Python 2.7.12

I grabbed the required gtk updates via ppa (`http://ppa.launchpad.net/gnome3-team/gnome3-staging/ubuntu xenial main`)
`dpkg -l | grep gtk-3` 
ii  gir1.2-gtk-3.0:amd64	3.20.8-1ubuntu0\~ppa1
ii  libgtk-3-0:amd64	3.20.8-1ubuntu0\~ppa1
ii  libgtk-3-bin	3.20.8-1ubuntu0\~ppa1
ii  libgtk-3-common	3.20.8-1ubuntu0\~ppa1

---
Okay your comments made me test this thoroughly and I got more input on this issue:

I cleaned all my .cache
Also the _pango_ errors did not happen again.

I restarted rafcon (in shell) and created a new fresh state machine (lets call it **new-test-sm** to make this explanation easier) (HS + ES with loggers), everything works fine, DEBUG CONSOLE shows prints, its filters :ballot_box_with_check: :ballot_box_with_check: work fine. 
Then I open and run an older state machine I copied over from another rafcon (v0.13.8?) (lets call this one **old-sm**)
:boom: :boom: :boom: it does not output anything to DEBUG CONSOLE! However it is logged in the shell.
When I now run **new-test-sm**, it also prints no longer to the DEBUG CONSOLE?!
Now here is the magic:
I close **old-sm** and suddenly the DEBUG CONSOLE refreshes and shows all the logger prints from **old-sm** and **new-test-sm** runs I just did!
... and this massive wall of errors pop up in shell <details><summary>Click to expand</summary>
```
Traceback (most recent call last):
  File "/home/testuser/.local/lib/python2.7/site-packages/gaphas/decorators.py", line 129, in async_wrapper
    func(*args, **kwargs)
  File "/home/testuser/.local/lib/python2.7/site-packages/gaphas/view.py", line 798, in update_bounding_box
    cr = self.get_window().cairo_create()
AttributeError: 'NoneType' object has no attribute 'cairo_create'
2020-01-15 16:35:05:  WARNING - rafcon.gui.controllers.graphical_editor_gaphas:  There is no view for state model Model of state: None
Traceback (most recent call last):
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/controllers/graphical_editor_gaphas.py", line 723, in set_focus_to_state_model
    state_size = self.view.editor.get_matrix_i2v(state_v).transform_distance(state_v.width, state_v.height)
AttributeError: 'NoneType' object has no attribute 'editor'
Traceback (most recent call last):
  File "/home/testuser/.local/lib/python2.7/site-packages/gtkmvc3/model_mt.py", line 77, in __idle_callback
    method(*args, **kwargs)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 198, in root_state_model_before_change
    self._send_root_state_notification(model, prop_name, info)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 329, in _send_root_state_notification
    self.state_machine._notify_method_before(self.state_machine, cause, (self.state_machine, ), info)
AttributeError: 'NoneType' object has no attribute '_notify_method_before'
Traceback (most recent call last):
  File "/home/testuser/.local/lib/python2.7/site-packages/gtkmvc3/model_mt.py", line 77, in __idle_callback
    method(*args, **kwargs)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 212, in root_state_model_after_change
    self._send_root_state_notification(model, prop_name, info)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 331, in _send_root_state_notification
    self.state_machine._notify_method_after(self.state_machine, cause, None, (self.state_machine, ), info)
AttributeError: 'NoneType' object has no attribute '_notify_method_after'
Traceback (most recent call last):
  File "/home/testuser/.local/lib/python2.7/site-packages/gtkmvc3/model_mt.py", line 77, in __idle_callback
    method(*args, **kwargs)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 198, in root_state_model_before_change
    self._send_root_state_notification(model, prop_name, info)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 329, in _send_root_state_notification
    self.state_machine._notify_method_before(self.state_machine, cause, (self.state_machine, ), info)
AttributeError: 'NoneType' object has no attribute '_notify_method_before'
Traceback (most recent call last):
  File "/home/testuser/.local/lib/python2.7/site-packages/gtkmvc3/model_mt.py", line 77, in __idle_callback
    method(*args, **kwargs)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 212, in root_state_model_after_change
    self._send_root_state_notification(model, prop_name, info)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 331, in _send_root_state_notification
    self.state_machine._notify_method_after(self.state_machine, cause, None, (self.state_machine, ), info)
AttributeError: 'NoneType' object has no attribute '_notify_method_after'
Traceback (most recent call last):
  File "/home/testuser/.local/lib/python2.7/site-packages/gtkmvc3/model_mt.py", line 77, in __idle_callback
    method(*args, **kwargs)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 198, in root_state_model_before_change
[... 40 lines cut ...]
    self._send_root_state_notification(model, prop_name, info)
[... cut ...]
    self._send_root_state_notification(model, prop_name, info)
  File "/home/testuser/.local/lib/python2.7/site-packages/rafcon/gui/models/state_machine.py", line 331, in _send_root_state_notification
    self.state_machine._notify_method_after(self.state_machine, cause, None, (self.state_machine, ), info)
AttributeError: 'NoneType' object has no attribute '_notify_method_after'
```
</details>

Not sure if related.
 But definitely something fishy is going on with the **old-sm** opened and it is affecting the rafcon GUI.

LJMP, 2020-01-20:
> Hi @LJMP, could you provide us with **old-sm** (zip it and upload it here)? It sounds like the issue comes from that state machine.

okay :+1:  I uploaded it into a git repo and invited you.
Hope you can find the issue :pray:

franzlst, 2020-01-24:
Hi @LJMP,

@sebastian-brunner tested your baked state machine on Ubuntu 16.04 with xfce and could not reproduce the error. Could you check, whether the error occurs in your setup with the baked state machine you provided to use? I could imagine that re-saving the state machine might have removed the issue.

sebastian-brunner, 2020-02-13:
As @franzlst wrote I could not reproduce the error, neither in 18.04, 16.04, gnome or xfce. I would assume that a broken environment is the reason for your bug.
For now I will close the issue. Feel free to reopen it if you have more information about the bug.

closed 2020-02-13 by sebastian-brunner