DLR-RM/RAFCON #32: Contributing guidelines not easily accessible

nickswalker, 2020-02-08:
GitHub displays lots of helpful prompts for new contributors, but clicking on them is likely to confuse users because they just point to the CONTRIBUTING symlink. Perhaps a compiled version of this file could be put there instead to improve the new contributor experience?
[image]
[image]
[image]

franzlst, 2020-02-12:
Hi @nickswalker, thank you for the hint! Unfortunately, GitHub cannot follow symlinks and only displays the path to the target document. This is indeed not very helpful for a user.

I don't want to have the same information twice in the repository, but maybe it would be an idea to turn the symlink around, i.e. have the actual document in the root and only a link in the documentation. I need to see, whether Sphinx can handle that.

franzlst, 2020-02-19:
Fixed in af2e26805e9ce7d8a1942270b087f224b9ef5e6b

closed 2020-02-19 by franzlst