DLR-RM/RAFCON #34: The DATA PORT is missing the “Scoped Variables” tab

lx852357, 2020-03-03:
My rafcon is v0.14.2 and my computer system is debian 9. 

While I following the tutorials, I cannot find the Scoped Variables tab. I see this tab in DATA PORT in the picture on your homepage. Hope to get answers. Thanks!

franzlst, 2020-03-03:
Hi @lx852357,
my first idea is that the state you have selected is an `ExecutionState`, which does not have `ScopedVariable`s. If you select a `HierarchyState`, is the tab also missing?

In order to help you better, a screenshot would be helpful. It would also be interesting, if there are any error message in the console.

lx852357, 2020-03-03:
Thank you! The tab appeared when I select a HierarchyState. 

Sorry, I have screenshot but I also cannot found the upload button.  Anyway, this problem has been solved. Thank you very much!

closed 2020-03-03 by lx852357