DLR-RM/RAFCON #36: How to use RAFCON for ROS2?

lx852357, 2020-03-19:
I created a execution state in root state.
execution state code:
```
import rclpy
from rclpy.node import Node

from std_msgs.msg import String

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            String,
            'topic',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

    def listener_callback(self, msg):
        self.get_logger().info('I heard: "%s"' % msg.data)

def execute(self, inputs, outputs, gvm):
    rclpy.init(args=args)

    minimal_subscriber = MinimalSubscriber()

    rclpy.spin(minimal_subscriber)

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_subscriber.destroy_node()
    rclpy.shutdown()
    return "success"
```

An error occurred while I clicked the apply button:

`13:21:29 ERROR - gui.controllers.state_editor.source_editor: The script was saved, but cannot be compiled: SyntaxError: invalid syntax (__init__.py, line 62)`

System: Debian 4.9.210
ROS2: eloquent

lx852357, 2020-03-19:
Besides, I have another question. 

If I use python script in ROS1. I don't need to compile by catkin_make. The script will work correctly as ROS node. 

But in ROS2. The script not compile by colcon build(contrast catkin_make). It will execute but cannot communication with another ros2 node. Unless you build it then using `ros2 run xxx` the python script would work correctly. I suspect because ROS2 used DDS.

And I think RAFCON project is not compiled with ROS2. 

Oh sorry, it may be the same problem. How to use RAFCON for ROS2?

franzlst, 2020-03-19:
Hi @lx852357,

I don't have any experience with ROS2, nevertheless I might be able to answer some of your questions.

The code you copied here uses Python 3 syntax (`super().__init__('minimal_subscriber')`, `super` with no arguments), this is why you need to run RAFCON with a Python 3 interpreter.

> And I think RAFCON project is not compiled with ROS2.

RAFCON is written in Python (supports both 2 and 3) and is thus interpreted (not compiled).

> How to use RAFCON for ROS2?

As I said, I don't know much about ROS2. One requirement (according to a quick Google search) is that ROS2 only supports Python 3, thus you need to run RAFCON with a Python 3 interpreter and write Python 3 compatible code. Running RAFCON with Python 3 requires you to install it with pip3!

I hope that helps!

lx852357, 2020-03-19:
Thank you @franzlst ,
Yes, It is interpreter nor compiler. 

I think ROS2 node may only using `ros2 run XX` this way will work correctly, otherwise the communication will not work.

So, I wanna know that can RACFON state machine project be part of my ROS2 project? And It's up to me to decide when to trigger the state machine process in my Python code.

lx852357, 2020-03-19:
> I think ROS2 node may only using ros2 run XX this way will work correctly, otherwise the communication will not work.

In ROS2 environment, executing the Python script directly it could work. It's my problem of environment.
Would you mind closing this issue later? There may be other problems in the future. It won't be long. I will close this question in this week. 
Thank you @franzlst .

lx852357, 2020-03-19:
Exciting! The RAFCON has successfully run ROS2 listener. 

But I still don't understand some details.

1. If I defined a global variable which is not saved in GVM. While the process has enter next state, can I still call the global varible?

2. Are local variable of previous state immediately destroyed after entering the next state.(Not concurrency state)

3. I have some custom message struct. How do I pass it to the next state? If not used GVM.

I used state machine to manage my process. But the data is global.
I want to see the data flow clearly.

franzlst, 2020-03-19:
> Exciting! The RAFCON has successfully run ROS2 listener.

Cool! Can you give some information, how you managed to do so now? Are there any special things others should be aware of? It would be great if you could share your experience!

Regarding the GVM questions: Please open a new issue and post them there!

Rbelder, 2020-03-19:
@franzlst I think he is doing simple **global** var. And you are right to request a ne issue. But maybe a short hint to our tutorials will help.

@lx852357 Using **global** statements is a very implicit way of holding permanent objects. Therefore RAFCON offers the GVM to do this transparent and secure (e.g. with locks). An example for a clean ros node creation you can find [here](https://github.com/DLR-RM/RAFCON/blob/master/share/rafcon/examples/libraries/ros_libraries/init_ros_node/init_ros_node_INIT_ROS_NODE)/script.py#L28. Please look also into our ROS turtle [tutorials](https://rafcon.readthedocs.io/en/latest/tutorials.html#starting-the-basic-turtle-demo-state-machine-using-ros) and there implementation to avoid redoing your work too often. Consider also our FAQ or other guides on [readthedocs](https://rafcon.readthedocs.io/en/latest/) in case you have not read them already.

lx852357, 2020-03-20:
Thank you!

I try @Rbelder opinion firstly. If I still have new questions I will open a new issue.

Some information about use RAFCON for ROS2:
1. **pip3** install rafcon --user

2. Because ROS2 natively supports distributed. So you can subscribe or publish directly. For example,
[image]

Code in the test state:
```
import rclpy
import sys
from rclpy.node import Node

from std_msgs.msg import String

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            String,
            'topic',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

    def listener_callback(self, msg):
        self.get_logger().info('I heard: "%s"' % msg.data)

def execute(self, inputs, outputs, gvm):
    rclpy.init(args=None)  # Here may report an error.  

    minimal_subscriber = MinimalSubscriber()
    minimal_subscriber.get_logger().info('execute')
    rclpy.spin(minimal_subscriber)

    minimal_subscriber.destroy_node()
    rclpy.shutdown()
```
3. An error may reported when running the state machine.
> RuntimeError: Failed to init: **rcl_init** called on an already initialized context

This problem occurs because of this subscriber may not end. Using `ros2 node list` to check if the subscribe still alive. When this problem occurred I have to shutdown RAFCON.

4. When running a publisher the ros logger will display on your terminal instead of RAFCON.

**By the way, I don't know if  the subscriber will end after stop a state machine.**

franzlst, 2020-03-20:
Thank you for the information!

> This problem occurs because of this subscriber may not end

I could imagine that this occurs when the state raises an exception for some reason and the `execute` function does not run to the end. Try this:

```python
try:
    rcpl.init()
    # more code
finally:
    rclpy.shutdown()
```

> By the way, I don't know if the subscriber will end after stop a state machine.

I don't know about ROS2, but I would imagine that `minimal_subscriber.destroy_node()` causes the subscriber to stop. This the subscribr would already end when the state finishes.

closed 2020-03-20 by lx852357