DLR-RM/RAFCON #37: Some question about the built-in data and variable in RAFCON

lx852357, 2020-03-20:
I have read the tutorials. But some questions still confuse me.

1. Are local variable of previous state immediately destroyed after entering the next state.(Not concurrency state)

2. I have some custom data struct. How do I pass it to the next state using data flow?  If I edit the `DataType` in `Data Linkage` to my custom data. RAFCON will report some error message like this,
[image]

3. Can I control the state machine to run or stop by code instead of the GUI?

franzlst, 2020-03-20:
> Are local variable of previous state immediately destroyed after entering the next state.

What do you mean with "local variable"? Input data ports or variables defined within `execute`?

> I have some custom data struct. How do I pass it to the next state using data flow?

The data strcut (I guess you mean a class) must be within a module that is within the `PYTHONPATH`. E.g.a class `Point` in the module `my_data_types`. Then the data type of e.g. an input data port would be `my_data_type.Point`.

> Can I control the state machine to run or stop by code instead of the GUI?

Yet, this is possible, but you will need to [write a plugin](https://rafcon.readthedocs.io/en/latest/plugins.html) for that. Some [example code](https://github.com/DLR-RM/RAFCON/blob/master/share/rafcon/examples/plugins/templates/core_template_observer.py) is shipped with RAFCON.

lx852357, 2020-03-20:
> What do you mean with "local variable"? Input data ports or variables defined within execute?

Yes, The "local variables" is defined within 'execute'.

```
def execute(self, inputs, outputs, gvm):
    local_variable = 10
    return 0
```
Whether the `local_variable` will be destroyed when entering next state? Further more, if I have a callback function will be called in previous state?

franzlst, 2020-03-20:
As soon as `execute` is left (and thus the state finishes), `local_variable` will be "destroyed". This is pure Python, the variable gets out of its scope.

> Further more, if I have a callback function will be called in previous state?

Please explain that a bit more detailed. Where is the callback function defined, where is it passed to, etc. The answer will again be independent of RAFCON, but purely Python related.

lx852357, 2020-03-20:
I think I already know the answer. This is a pure Python related question. I thought RAFON would manage the life cycle of each state before. Thanks for your answer！

closed 2020-03-23 by lx852357