DLR-RM/RAFCON #38: How to set default value of custom data type?

lx852357, 2020-03-23:
I create a custom data type like this:
```
import json

class Point:
  def __init__(self,px=0,py=0):
    self.x = px
    self.y = py

  def toJSON(self):
    return json.dumps(self,default=lambda o:o.__dict__,sort_keys=True,indent=4)
```

**State source:**
```
def execute(self, inputs, outputs, gvm):
    point = inputs['point']
    self.logger.info("[{}, {}]".format(point.x,point.y))
    return 0
```
**Input Ports:**
[image]

**Console output:**
[image]

The default value will be directly mapped to point.x. 
When I set "`[1,0]`" as default value the output is "`INFO - test: [[1,0], 0]`". The same for others.

sebastian-brunner, 2020-03-24:
Hi @lx852357, 
currently we don't support the setting of default values for custom data types, only primitive data types such as float, list, dict, string, tuple etc.

You can create a small generator state which creates the data as you want.

sebastian-brunner, 2020-03-24:
An enhancement on our side would be to accept e.g. a list for the default values and create an object of the given class by using the elements in that list for the constructor parameters.

franzlst, 2020-03-24:
Well, there is at least _rudimentary_ support for parsing default values into custom types. Have a look [at the code responsible for the conversion](https://github.com/DLR-RM/RAFCON/blob/master/source/rafcon/utils/type_helpers.py#L73). In case of a custom class, the function tries to call the constructor of the class with the default value passed as string. Therefore, you can do something like this:

```python
import json

class Point:
  def __init__(self,px=0,py=0):
    if isinstance(px, str):
        self.x, self.y = my_str2point_parser(px)
    else:
        self.x = px
        self.y = py
```

Note: I have never tested this!

franzlst, 2020-03-24:
Did it work?

lx852357, 2020-03-30:
Yes, it worked correctly！
```
import json
class Point:
    def __init__(self,px=0,py=0):
        if isinstance(px,str):
            self.x,self.y = self.str2point_parser(px)
        else:
            self.x = px
            self.y = py
    def toJSON(self):
        return json.dumps(self, default=lambda o: o.__dict__, 
            sort_keys=True, indent=4)

    def str2point_parser(self,px): # “10，100”
        point = px.split(",")
        return int(point[0]), int(point[1])

```

franzlst, 2020-03-30:
Cool, thx!

closed 2020-03-30 by franzlst

sebastian-brunner, 2020-03-30:
@franzlst: nice idea btw.! At least, this works for data types were you have own control of (i.e. not for default ros types).