DLR-RM/RAFCON #39: Successfully loaded plugin but not available.

lx852357, 2020-03-24:
My plugin path: `/home/bot/rafcon_plugins/templates`.(download from this repo/master)
The fold like this:
[image]

While set the **RAFCON_PLUGIN_PATH**=`/home/bot/rafcon_plugins/templates`.

An error reported:`ERROR - rafcon.utils.plugins:  The specified plugin path does not exist: /home/bot/rafcon_plugins/template`.

When set **RAFCON_PLUGIN_PATH**=`/home/bot/rafcon_plugins/` the message is:
 `2020-03-24 16:16:42:     INFO - rafcon.utils.plugins:  Found plugin 'rafcon_plugins' at /home/bot/rafcon_plugins/`
`2020-03-24 16:16:42:     INFO - rafcon.utils.plugins:  Successfully loaded plugin 'rafcon_plugins'
`

But it does't call hooks.py because I have not seen this [message](https://github.com/DLR-RM/RAFCON/blob/master/share/rafcon/examples/plugins/templates/hooks.py#L15).

And the tutorial said put `xxx/plugins/templates` into **RAFCON_PLUGIN_PATH** instead of `xxx/plugins`.
[image]

Besides, I found [that](https://github.com/DLR-RM/RAFCON/blob/master/source/rafcon/utils/plugins.py#L43).

I have a test.

**Test code:**
```
import os 

path1 = '/home/bot/rafcon_plugins/templates'
isExist = os.path.exists(path) 
print(isExist) 
  
path2 = '/home/bot/rafcon_plugins/templates/hooks.py'
isExist = os.path.exists(path) 
print(isExist) 

path2 = '/home/bot/rafcon_plugins'
isExist = os.path.exists(path) 
print(isExist) 
```
Outputs:
```
False
True
True
```

Is something wrong with me?

franzlst, 2020-03-24:
`RAFCON_PLUGIN_PATH` should be set to `/home/bot/rafcon_plugins/templates`. If you didn't make a cop'n'paste error, then you forget the 's':
`ERROR - rafcon.utils.plugins: The specified plugin path does not exist: /home/bot/rafcon_plugins/template` <= No 's' at the end

lx852357, 2020-03-30:
Sorry, it is my fault.......

closed 2020-03-30 by lx852357