DLR-RM/RAFCON #220: New states have to be inserted at current mouse position

franzlst, 2020-04-17:
Right now if a new state is created, it is created at the next position of the virtual grid of the parent state.
This new position can be hehind an existing lager state. So the new state will be completely invisible.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2016-11-08 17:06:33+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
I would like to back port the feature then to open gl, too. I comment so that nobody is confused. Also the idea came up already at times when we only have had open gl.

Added at 2017-02-24:
This issue needs a representation (test) in issue #175 to preserve gaphas functionality and as reliability to the user.

Originally created by @Rbelder  at 2016-11-09 10:07:59+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-12-13:
Seems implemented already

closed 2024-12-13 by JohannesErnst