DLR-RM/RAFCON #351: Show library usage

franzlst, 2020-04-17:
Often library states are used across the whole workspace. While refactoring a library it is important to know where the library is included. Thus a dialog which shows all usages of a library state would be usefull.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2017-05-24 05:41:10+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
What about using the State Machine Tree for this? Either in its current or an extended version?

The tree show all states. If the focus is in the widget, you can start typing and it will jump to entries starting with these letters. Currently, only the first column is searched. Maybe one can extend this to search in all columns. Maybe it can also be extended so that the search works as filter. If then a further column is inserted, showing the library name, I guess its all you need :smile: 

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2017-05-24 06:36:31+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
I think Sebastian means a kind of feature like the "Find Usages" in PyCharm which can be very use full. 

From my point of view a the library manager needs to index it libraries what could be done in a separate thread and the state machines that are open or get open do a similar index for there self. In any case this index can be improved in the future and could be integrated into a search widget (to be effective). Therefore see also the discussion in issue #35 the feature could be integrated in such a widget/class.

In general, I think it is more efficient to start directly a integration of a search widget which can visualize the results of such a search instate creating some kind of dialog. The search results widget can be situated in the lower bar similar to PyCharm. This widget also can be used to modify the selection and of library tree, state machines manger and state machines.

Originally created by @Rbelder  at 2017-05-24 08:22:29+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Yes Rico got my point. The idea of creating a dedicated search window is even better. I did not think about such a feature in the first place but I use it a lot in PyCharm. 

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2017-05-24 11:46:46+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-02-02:
Implemented in 0.14.11

closed 2024-02-02 by JohannesErnst