DLR-RM/RAFCON #373: Support for rename libraries, library paths and library root names

franzlst, 2020-04-17:
The idea is that the core and/or GUI helps to rename of folders (library path), library root keys and library names and adapt all related open state machines and mounted libraries (that employ this changed element) according this change. 

There are basically three levels of implementations for this feature:
1. add rename functionality for all items (library name, library path elements which also include the library root key and needs adaptation of the core configuration)
2. rename all used instances of this library in open state machines (set the dirty flag of the state machine)
3. update all library state machines library states that employ the renamed element 
    * which is dependent to issue #371 because otherwise you have to implement loop opening of all library state machines
    * additional issue #372 will support this kind of feature

The issue is helpful to also correct bad naming of folders.

It can be supported by implement and index of issue #372 for LibraryState usage to find very fast instances of a library without looping state machines and so on. Therewith we also can detect influence of changed libraries onto open state machines.

Add your sub-tasks and your opinion about the feature. The feature firstable will be only partly implemented because sub task 3 is quit complex and can come up with problems not pointed out in respective issue, till now.

Originally created by @Rbelder  at 2017-07-12 13:24:34+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Upvote! At least level 1.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2018-07-19 06:52:29+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-12-13:
This was implemented

closed 2024-12-13 by JohannesErnst