DLR-RM/RAFCON #420: Console output should handle high frequency messages

franzlst, 2020-04-17:
Console widget stops to show console output from a output load > approx. 50 logs per second.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2017-10-12 09:24:39+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Eventually a flag for each state if logging is desired would be nice .... the flag should also be stored in the meta_data and restored after loading.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2017-10-12 09:34:02+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Ok in our case, most of the output is generated by state transitions and global variable access (we have parallel branches operating at the same time). ... so we should try that 50 logs per seconds are fine for the console widget.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2017-10-12 09:35:36+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
I think the LoggingView shows all log messages, also for higher frequencies, but (very) belated. The reason is that the messages are printed using `idle_add` with a low frequency to prevent the GUI from freezing. I guess the methods could be optimized a little bit, but else I don't see a better solution concerning the widget.
Besides that, a filtering option would definitely be helpful, e.g. by namespace.

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2017-10-16 12:11:10+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Proposed solution from 2017-12-04:

Hide debug messages, if GUI is busy (deactivate checkbox after certain threshold):

`idle_add` is used to show logs in the LoggingView. A separate Thread (or something similar) checks, whether the log message is shown within a certain threshold. If this is not the case, the `debug` message are being deactivated. If the user later activates the level again, the view is being refreshed.

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2017-12-05 15:11:22+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-12-13:
Improved since https://github.com/DLR-RM/RAFCON/releases/tag/2.0.0

closed 2024-12-13 by JohannesErnst