DLR-RM/RAFCON #542: search for all library usages

franzlst, 2020-04-17:
In common IDEs the user can search for the usages of functions. This would also be nice for libraries in RAFCON.

This is a low priority issue, and open for discussion.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2018-05-29 05:28:49+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Based on this the user could replace the occurrence of one library with another library having the same interface (i.e. input and output ports).

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2018-05-29 05:29:29+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Yes that issue we have had already very early #35 and and later again #351. And we discussed already what kind of features we could full-fill with a proper search and replace widget.

Originally created by @Rbelder  at 2018-05-29 14:29:19+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
I created a project "Search Widget" to track our efforts for this kind of features and there discussion.
 

Originally created by @Rbelder  at 2018-05-30 14:17:56+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-02-02:
Implemented in 0.14.11

closed 2024-02-02 by JohannesErnst