DLR-RM/RAFCON #554: child state cannot be moved anymore

franzlst, 2020-04-17:
Traceback (most recent call last):
  File "/internal_path/common/foreign_packages/python-gaphas/0.7.2/osl42-x86_64/lib/python2.7/site-packages/gaphas/view.py", line 863, in do_event
    return self._tool.handle(event) and True or False
  File "/internal_path/common/packages/rafcon/0.12.12/source/rafcon/gui/mygaphas/tools.py", line 55, in handle
    rt = super(ToolChain, self).handle(event)
  File "/internal_path/common/foreign_packages/python-gaphas/0.7.2/osl42-x86_64/lib/python2.7/site-packages/gaphas/tool.py", line 208, in handle
    return self._grabbed_tool.handle(event)
  File "/internal_path/common/foreign_packages/python-gaphas/0.7.2/osl42-x86_64/lib/python2.7/site-packages/gaphas/tool.py", line 126, in handle
    return self._dispatch(event)
  File "/internal_path/common/foreign_packages/python-gaphas/0.7.2/osl42-x86_64/lib/python2.7/site-packages/gaphas/tool.py", line 121, in _dispatch
    return bool(h(event))
  File "/internal_path/common/packages/rafcon/0.12.12/source/rafcon/gui/mygaphas/tools.py", line 168, in on_button_release
    inmotion.move((event.x, event.y))
  File "/internal_path/common/packages/rafcon/0.12.12/source/rafcon/gui/mygaphas/guide.py", line 90, in move
    super(GuidedStateInMotion, self).move(pos)
  File "/internal_path/common/foreign_packages/python-gaphas/0.7.2/osl42-x86_64/lib/python2.7/site-packages/gaphas/guide.py", line 219, in move
    dx, edges_x = self.find_vertical_guides(item_vedges, pdx, h, excluded_items)
  File "/internal_path/common/packages/rafcon/0.12.12/source/rafcon/gui/mygaphas/guide.py", line 30, in find_vertical_guides
    if not self.item.parent:
  File "/internal_path/common/packages/rafcon/0.12.12/source/rafcon/gui/mygaphas/items/state.py", line 268, in parent
    return self.canvas.get_parent(self)
AttributeError: 'NoneType' object has no attribute 'get_parent'

Originally created by doem-an  at 2018-06-09 11:34:49+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
I guess without information what has been done before to reach this exception (e.g. copy/paste, group, substitute, etc), we won't be able to fix this.

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2018-06-11 08:10:37+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
@doem-an Yes, @franzlst is right can you reproduce by storing the state machine and doing this. Also it is of interest what is written in the log before this traceback occur. You can also load up the baked state machine here if you can reproduce it in the stored and reloaded state machine. Thanks.

@sebastian-brunner and @franzlst we should do the how i call it the "bake session" feature where we bake not only the state machine but also at least the history and log.

Originally created by @Rbelder  at 2018-06-11 08:59:41+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-12-13:
cannot reproduce

closed 2024-12-13 by JohannesErnst