DLR-RM/RAFCON #596: External editor does not remember the handed command and also does not lock the embedded editor

franzlst, 2020-04-17:
So by default the external editor is not set:
- after the dialog open to request a command to open a respective editor the command is not remembered (if check box is enabled) in respective gui_config.yaml
- the respective embedded editor is not locking it self after the external editor has been open

Because the editor is not locked after the external editor is closed the unlock button is not available to reload the changes properly.

As long as the editor is working (behavior as assumed) if respective command is set by the preference window or directly in the gui_config.yaml file this issue gets also the low priority label.

Originally created by @Rbelder  at 2018-09-03 13:32:08+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Optionally, make use of a GTK+ 3 [AppChooser](https://lazka.github.io/pgi-docs/#Gtk-3.0/classes/AppChooser.html#Gtk.AppChooser)

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2018-10-15 13:46:12+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Funny failure, too, the d9c91da hast the last line `This solve #596` but issue was not closed automatically.

Originally created by @Rbelder  at 2018-12-05 12:27:39+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
`solve` is not [one of the keywords](https://help.github.com/articles/closing-issues-using-keywords/)

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2018-12-05 12:33:08+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Issue needs to be reopen, it is not working again. This time the Dialog is shown but the text field is missing. So no command can be added.

This Time at test will be needed to close the issue.

Originally created by @Rbelder  at 2019-02-12 15:59:52+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Maybe the issue is a little bit different because the field is not accessible but still the editing of the command is not possible at the moment.

Originally created by @Rbelder  at 2019-02-12 16:05:26+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Could you please open a new issue for this?

First of all, it is not related to the previous issue (the remembering of the command). Second, we agreed upon creating all new issues on GitHub (not enterprise) from now on.

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-02-12 16:08:03+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Yes please issue on public github.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2019-02-18 08:29:47+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-12-13:
Cannot reproduce

closed 2024-12-13 by JohannesErnst