DLR-RM/RAFCON #613: Different colors for different states

franzlst, 2020-04-17:
Different state should have different colors.

Also customizable colors would be nice.

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2018-10-11 08:32:23+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
This is quite nice idea to help the user to get a quick overview of a state machine. The implementation should be cheap, too. At least it is cheaper then having complete different views for different types and it is much more transparent.

So maybe we use the issue to discuss reasonable colors and those which we should not take like, `green`, `blue`, `yellow` which are already taken. 

Maybe the background color would be the best target. In this way we also can reuse already used colors with transparency.

Originally created by @Rbelder  at 2018-10-11 09:00:18+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-12-13:
Already implemented

closed 2024-12-13 by JohannesErnst