DLR-RM/RAFCON #708: Jenkins feature wish list

franzlst, 2020-04-17:
This is what I plan to add to Jenkins somewhen in the future:
* [x] pytest coverage report
  * this should be a no-brainer, but it will extend the times a build runs (therefore, see next point)
* [x] parallel run of tests
  * this will probably not only require adaptions to the tox/Jenkinsfile, but also more fundamentally to RAFCON. Two RAFCON instances must not interfere with each other.
* [ ] integrate [SLOCCount Plugin](https://wiki.jenkins.io/display/JENKINS/SLOCCount+Plugin) (counts lines of code)
  * shouldn't be much work
* [x] Collect pytest warnings (similar as it is now done for sphinx warnings)
  * here, I only wait for @crem-ja to implement some requested changes
* [ ] add further steps of [release steps](https://rafcon.readthedocs.io/en/latest/development/development.html#steps-for-releasing)
  * for this, I must find out how to push changes back to repo from Jenkins. I found [this thread](https://issues.jenkins-ci.org/browse/JENKINS-28335) with possible solutions.

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-03-18 12:37:37+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
See:
https://rmc-github-server/tools/cissy/issues

Originally created by crem-ja  (Jan.Cremer@dlr.de) at 2019-03-18 12:58:03+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Pytest warnings are [now](https://rmc-github-server/common/rafcon/commit/fad35496) parsed, too.

See [this build result](https://rmc-jenkins.robotic.dlr.de/jenkins/job/common/job/rafcon/job/jenkins/154/pytest/) for an example.

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-03-18 15:04:29+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Coverage report in pull request #746 

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-06-24 10:47:38+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Parallel test runs in pull request #754

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-07-11 11:13:28+00:00 (moved from RMC internal repository)

JohannesErnst, 2024-12-13:
Closed due to inactivity

closed 2024-12-13 by JohannesErnst