DLR-RM/RAFCON #709: Repetitive run of generic dialog MessageDialog crash`s RAFCON

franzlst, 2020-04-17:
I think the dialogs are not tested hard enough because one at least is not robust. 

I would extend respective test with kind of loop that runs at least this dialog (MessageDialog) multiple times and then debug the dialog MessageDialog.

As long as we generally do not use this dialog, I have not labeled it critical but feel free to do so.

Originally created by @Rbelder  at 2019-03-22 15:09:03+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Why do you think the dialogs are buggy? What did you do and what did happen...?

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-03-25 07:49:36+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Sorry but see above again. 

So I wrote ONE dialog the MessageDialog is buggy, see above.
I also wrote that I run the MessageDialog repetitively and it crash`s the the RAFCON Instance.

Sorry for the two typos in the headlines but humans generally are failure tolerant beings.

Can you please confirm that it crashs in your case, too.

Originally created by @Rbelder  at 2019-03-25 09:46:46+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
> Sorry for the two typos in the headlines but humans generally are failure tolerant beings.

Please read your texts a second time before submitting. There are often types/errors in your texts that make them hard to read. (I really try hard!)

It would also help, if you add more information. What does "repetitive" mean? Two times? 100 times? With pause in between?

I just created a state machine with a root state and `MessageDialog` as only child state, which outcome is connected back to the income. I ran the state machine and confirmed the dialog a dozen times, no crash occurred.

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-03-25 12:52:05+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
So dozen times was never needed, in my case, while the hand full of runs I made. I assume the failure occurs in my case after 5 to 10 runs.

The only difference was that I run 5 to 10 times the root-state with this MessageDialog in. So the hole state machine. But I think this while not change the outcome in your case.

I will try hard to improve my texts. I wrote the issue 5 mins before I left the institute. And we have the idea to write the issue instate of leaving it. So please stay tolerant if someone writes a minimal issue. We are friends.

Originally created by @Rbelder  at 2019-03-25 13:11:13+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
> The only difference was that I run 5 to 10 times the root-state with this MessageDialog in.

So you executed the state machine with that `MessageDialog` library as child state several times?

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-03-25 13:17:21+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
> >    The only difference was that I run 5 to 10 times the root-state **with this MessageDialog in**.

> So you executed the state machine with that MessageDialog library as child state several times?

With this MessageDialog in the root-state not as the root-state.

So the same as you have done:
- new state machine
- drag and drop library MessageDialog in
- maybe change the text
- run it by F5  5 to 10 times.

Originally created by @Rbelder  at 2019-03-25 13:21:02+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
The dialogs were often unstable in the past. However, I cannot reproduce this one. I ran the state machine 20 times by just pressing F5 and Enter quickly but there was no error.
Could you provide more detail how I can reproduce this?

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2019-05-09 06:37:28+00:00 (moved from RMC internal repository)

sebastian-brunner, 2021-11-17:
Close non-reproducible issue.

closed 2021-11-17 by sebastian-brunner