DLR-RM/RAFCON #755: Positions of ports are not always updated

franzlst, 2020-04-17:
When changing the State machine model of a state machine, opened in the Gui, e.g. layouting it, the states within the machine are automatically rearranged, but the in / outcomes are not. 
The effect on the in / outcomes is not visible until the state machine was refreshed.

Originally created by @CSuerig  at 2019-07-05 08:37:00+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
@CSuerig: Could you please verify that the issue is fixed (use the develop branch)

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-07-09 06:03:51+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
@franzlst: As far as i can tell, the bug isn't fully fixed yet. I can observe, that only outcomes are updated now, but no incomes.

Originally created by @CSuerig  at 2019-07-12 08:22:45+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
@CSuerig: I could reproduce the issue when undoing move operations using the history. I fixed it in the [`fix/755/update_position_of_incomes` branch](https://rmc-github-server/common/rafcon/tree/fix/755/update_position_of_incomes). Could you please check again, if everything is working now? If it is not, please tell me, how I can reproduce the erroneous behaviour.

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-07-15 15:01:29+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
@franzlst: Unfortunately the bug persists. Inside of the `fix/755/update_position_of_incomes` Branch neither the incomes, nor the outcomes are updated. To be honest i am not familiar enough with Rafcon to know a way how to reproduce the erroneous behaviour, except using the Task Planner Plugin. 

If you want to reproduce it using the Task Planner Plugin, here are Steps to do so: 
To be able to generate a State machine in a relative short amount of time, you can nearly follow the [Turtle Sim Example](https://rmc-github-server/moro/rafcon_task_planner_plugin/blob/develop/examples/turtle_sim_example/turtle_sim_example.md).  
  
Nearly follow, because you have to make the following changes:   
- you have to use the Branch develop
- Before Clicking on "Plan Task": 
     - First make a new state machine (With an empty Hierarchy State)
     - Open the Semantic Data Section, and change the value of "Allow_override" in "RAFCONTPP" to "True". (that's necessary to use it later again for planning without having to clear it manually) 
    - Save the State machine, and select the Hierarchy State 
- After Clicking on "Plan Task":
   - Choose "selected State" in the field "Generate State machine Into".

Then you can Generate the state machine, and the bug will become visible. If you want to repeat the experiment, you just have to select the state again, click "Plan Task" and "Generate State machine", all your configuration should have been stored.

Originally created by @CSuerig  at 2019-07-19 06:54:12+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
I had a look at [your layouter code](https://rmc-github-server/moro/rafcon_task_planner_plugin/blob/develop/source/rafcontpp/logic/state_machine_layouter.py), which I guess is the code your are using to update the port positions.

I think the reason, why the graphical editor is not updating the ports is that is is never told that the meta data was updated. You can do this e.g. with the following code:

```python
msg = MetaSignalMsg('Layouter', "all", True)
root_state_m.meta_signal.emit(msg)
```

Some notes regarding the code:
* The OpenGL editor has been removed from RAFCON, so there is no more need for meta data for this editor
* there is a helper function to set meta data for a model, e.g. `root_state_m.income.set_meta_data_editor("rel_pos", (2, 80))` 

Originally created by @franzlst  (Franz.Steinmetz@dlr.de) at 2019-07-19 12:29:25+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
Thanks, that's a good Hint, i will remove the OpenGL code, and add the helper functions! 

After calling the Layouter, i emit this signal: 
```
target_state_m.action_signal.emit(ActionSignalMsg(action='substitute_state', origin='model',
                                                                action_parent_m=target_state_m,
                                                                affected_models=[target_state_m], after=True))
```

I tried your solution, by emitting  the meta_signal, but no success.

Originally created by @CSuerig  at 2019-07-19 13:37:08+00:00 (moved from RMC internal repository)

franzlst, 2020-04-17:
We have to be able to reproduce this inside RAFCON with as much overhead as possible! Currently the steps to reproduce are too many!

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2019-08-06 11:44:28+00:00 (moved from RMC internal repository)

CSuerig, 2020-04-23:
The issue is already fixed. Please close.

closed 2020-04-24 by franzlst