DLR-RM/RAFCON #763: Add documentation about autolayout plugin

franzlst, 2020-04-17:
Add documentation about how to use the autolayout plugin:

https://rmc-github-server/moro/rafcon_auto_layout_plugin

Originally created by @sebastian-brunner  (Sebastian.Brunner@dlr.de) at 2019-07-23 06:32:54+00:00 (moved from RMC internal repository)

ljklonepiece, 2022-04-13:
@franzlst  hi, may i know if there is any public doc about autolayout?

franzlst, 2022-04-19:
@ljklonepiece sorry, unfortunately, there is no further information. I think we also have never made this plugin open source. @sebastian-brunner do you know more?

sebastian-brunner, 2022-04-19:
@sebastian-brunner : As far as I know it was not yet released under an open-source license by DLR. But it would be awesome if DLR would do so!

Rbelder, 2022-04-20:
Hello all, 

most likely the plugin was never finished in a manner to publish it. I can not find a version online that I can access. 

At the time I went on from DLR I wanted to clean it up to get it published and you both used it for an auto-translation project. Most likely it is still at the stage of not fully cleaned up and maybe it is also not using the proper way to use the plugin interface and no external property files.

ljklonepiece, 2022-04-22:
thank you for the update!
just wanna double check with you guys,
does the autolayout function aim to achieve the same as what [OGDF  ](https://ogdf.uos.de/) or [graphviz ](https://graphviz.org/) can do?
if so, we'd like to know how we can contribute to the release of this much needed feature

JohannesErnst, 2024-12-13:
Currently nobody is working on this anymore but the current auto-layout functionality needs to be reworked. Converting to discussion