DLR-RM/RAFCON #838: Question: How do I load a state machine non-interactively?

nickswalker, 2020-06-11:
I'd like to write a unit test that checks whether a state machine can be loaded. The outline of how I'm doing this is:

````
from rafcon.core.start import open_state_machine, setup_configuration
setup_configuration(None)
sm = open_state_machine(<path_to_my_state_machine>)
# Code that recursively checks for 'LIBRARY NOT FOUND DUMMY STATE`
````

This will make it easier to catch breaking changes to library states. However, I cannot figure out how to load a state machine without hitting an `input` that waits for the user to acknowledge any broken library imports:

https://github.com/DLR-RM/RAFCON/blob/e6ba1626723c8f6f4da7e336e1c5b0f20f12ab54/source/rafcon/core/library_manager.py#L272

This line gets tripped when the `LibraryState` constructor calls into to get the library path. That constructor has user interaction enabled by default. I guess that default makes sense since the framework is oriented towards use with the GUI, but I can't find a simple way to disable this.

franzlst, 2020-06-12:
Hi Nick,

I'm happy to hear that you are still using RAFCON :-)

I just had a look into the code and as you say, there seems to be no easy was to override the default value, due to the bad design.

However, I think I have found a workaround for your case. If `allow_user_interaction` is `True` and a library cannot be found, two functions in the `interface` module are called. This module is intended to be mocked (in fact, the `gui` module overrides the fuctions provided by the `core` module). So all you need to do is to patch `show_notice_func` and `open_folder_func` with two empty functions returning `None`.

This should the cause the `'LIBRARY NOT FOUND DUMMY STATE` states to be created.

nickswalker, 2020-07-16:
Ah, that does it. Thanks for the tip!

```
import rafcon.core.interface

rafcon.core.interface.show_notice_func = lambda *args, **kwargs: None
rafcon.core.interface.open_folder_func = lambda *args, **kwargs: None
```

closed 2020-07-16 by nickswalker