DLR-RM/RAFCON #839: History states in RAFCON

rohith-tantry, 2020-07-24:
As far as I can see there is no implementation of History state in RAFCON. Is there a workaround or any way to implement history states?

franzlst, 2020-07-24:
Hi @rohith-tantry, what is your definition of a history state? Are you talking about this here: https://help.anylogic.com/index.jsp?topic=%2Fcom.anylogic.help%2Fhtml%2Fstatecharts%2FHistory+State.html?

Rohith-Sewts, 2020-07-24:
Thanks for the swift reply @franzlst . Yes, my query is to implement shallow and deep history as mentioned in the link.

sebastian-brunner, 2020-07-26:
tldr: No, for good reasons, RAFCON does not support history states. In general you would use one or several decider states that, based on the data you aggregated in the data-flows or in a knowledge base, decide upon the next state to execute.

Actually we could support them easily as we already hold a very detailed history of a state machine execution. You could simply retrieve the last executed child of a container state with a few lines of code. And then you would need to setup a dedicated state that retrieves this information and activates the respective child.

However, when designing RAFCON (https://elib.dlr.de/112067/2/elib.pdf) we deliberatively decided against supporting history states. They are an anti-pattern for complex behavior: they allow the state machine execution to jump do different states in a more or less uncontrolled manner. The main idea behind RAFCON was to model the control and data flow explicitly and offer the developer nice introspection tools. We leverage many advantages commonly known from graphical programming and dataflow programming. RAFCON was also strongly influenced by flow-charts were you can clearly retrace where which decisions were taken. History states destroy these benefits as the control flow is not bounded to transitions any more. You loose the explainability and traceability you tried to gain by using such a tool.

History states definitely help you to express medium-sized problems more elegantly such that you can save a few condition states. However, for complex problems (we programmed state-machines with several hundreds and thousands of states) it simply does not scale. I have some experience with boost state-chart: In your 7 layers deep state machine some state is suddenly executed because some history state stored some state reference sometime ago.

Apart from that, deep history states destroy any kind of modularization in your code because state activations just ignore the container boundaries. How are you going to develop reusable functionality that you want to offer via libraries to other developers?

History states are also bad for proper semantic modelling: one history state can aggregate a multitude of different execution traces that you should model explicitly with proper semantics!

If you are not sure how to model your use case without history states just give us some insights about your scenario. Then we can try to help you.

Rohith-Sewts, 2020-07-27:
Thankyou @sebastian-brunner . The argument seems compelling. As per my use case , say my manipulator is in some state carrying out a manipulation task and due to faulty socket connection between the plc and the robot the communication is lost and we have to manually do an emergency stop. But after the socket connection is restored back the robot should continue from the state as it was in , before hitting the emergency stop. This is where I have the need or rather feel that I need  the history sate.

Rohith-Sewts, 2020-07-27:
Also if my state machine is interrupted with an external event in the middle of carrying out a task in a pipeline, I would not want to start the whole pipeline from the beginning but proceed with the last interrupted state upon restarting the state machine.

franzlst, 2020-07-28:
I think what you describe are typical examples you have tackle when programming robots. I think @sebastian-brunner can share with you here with some experience with exactly the same issues!

sebastian-brunner, 2020-07-28:
Thx, for sharing your use case.

Usually, if you do an emergency stop then the robot did something you did not expected. Thus all states in your original container state (that you might to start from using a history state) won't do anything sensible as the assumptions you relied upon don't hold any more. Normally, after an emergency stop you need an error handling procedure. There are different possibilities, please find information [here](https://rafcon.readthedocs.io/en/latest/faq.html?highlight=error#how-to-handle-a-state-abortion-correctly) or in our publications (see above's link and [here](https://elib.dlr.de/121511/1/elib.pdf)). After the error handling you then analyze your environment and trigger a procedure than brings your robot back to a valid context. Finally, you would then be able to start from dedicated re-entry points. These re-entry points are triggered depending on the robot's belief state, which should reflect all changes the robot performed before it was preempted. However, trying to start from an arbitrary state (by using a history state) won't work for any non-trivial case.

Concerning communication delays or blackouts: Try to handle such low level things in the functional layer and not in the procedural layer (also often called the [procedural layer](https://www-robotics.jpl.nasa.gov/publications/Issa_Nesnas/01_volpe_claraty_aerospace.pdf)), which is what RAFCON is intended for.

> Also if my state machine is interrupted with an external event in the middle of carrying out a task in a pipeline, I would not want to start the whole pipeline from the beginning but proceed with the last interrupted state upon restarting the state machine.

I never said, you should start from the beginning. Just use proper re-entry points that can be selected by decider/condition states that leverage the robot's belief state. A single (or a few) decider states are able to handle a broad set of cases. If you end up having too many cases think about [integrating a planner](https://github.com/DLR-RM/rafcon-task-planner-plugin) and only use RAFCON for modelling the robot skills.

sebastian-brunner, 2021-11-17:
Closing, as of inactivity. Please reopen again if required.

closed 2021-11-17 by sebastian-brunner