DLR-RM/RAFCON #842: RAFCON Crashes abruptly.....

Rohith-Sewts, 2020-09-15:
I have a state machine built using RAFCON for a robot PLC system at our company. I am testing the state machine live with the PLC where the state machine running on a system communicates with the  PLC through a socket connection. The attached image shows the PLC part for the state machine. 
[image]
In the image the KEVIN(PLC) is a preemptive concurrency state which runs four child states handling the TCP socket connection with the PLC. These states are long running states and I have implemented the with proper preemption and abortion handling. It runs fine for a some time, but the GUI hangs abruptly, some times even RAFCON crashes. 
Since this part is very criticial at our company and since I also worked at DLR before(during my master thesis) and worked with RAFCON a  bit I proposed implementing the state machine at our company with RAFCON. 

The states run quite nice in the background and gives me a stable socket connection, but RAFCON GUI CRASHES...may be issue with GTK and Multithreading....Urgent help appreciated.

Rohith-Sewts, 2020-09-15:
This is the error with the crash....
Pango:ERROR:/build/pango1.0-Ne8X8r/pango1.0-1.40.1/./pango/pango-glyph-item.c:319:pango_glyph_item_iter_next_cluster: assertion failed: (iter->start_char < iter->end_char)

franzlst, 2020-09-15:
Hi @Rohith-Sewts, could it be related to this bug: https://gitlab.gnome.org/GNOME/pango/-/issues/346

franzlst, 2020-09-17:
One more question: do you dynamically change names of states (`self.name = newname`) or ports? I am asking because Pango is only involved for rendering the labels of states and ports.

Rohith-Sewts, 2020-09-18:
I just reinstalled rafcon and now it get no such crashes..

franzlst, 2020-09-18:
I'm happy to hear that you issue is gone! Maybe some dependency (cairo, pango, etc.) got updated meanwhile.

closed 2020-09-18 by franzlst

Rohith-Sewts, 2020-09-18:
I had one more query..I have 8 subscribers running in parallel . I run an init node while starting the state machine. When I get a flag form one of the subscribers I return preempted outcome which preempts the other parallel subscribers which works as I intend. But in the background the subscribers still run although the states are nor active which is because the node is still active. So I kill the node on exit but when I renter the state machine i cant initialize a new node since ROS allows to initialize node only once in one process..Is there a work around for this where I can kill the subscribers without killing the node.....

franzlst, 2020-09-18:
I think this question is more related to ROS rather than RAFCON. I haven't worked with ROS for a long time, so I cannot really give you an answer on this (maybe @sebastian-brunner can). You would need to find a way to stop the subscribers in a state that is connected to the preempted outcome of each of the states. I'm sure there is a way in ROS to stop a subscriber. It is also possible to handle a preemption "query" from inside a state, see [this](https://rafcon.readthedocs.io/en/latest/faq.html#how-does-preemption-work-how-do-i-implement-preemptable-states-correctly) and [this](https://rafcon.readthedocs.io/en/latest/faq.html#what-happens-if-a-state-machine-is-paused-how-can-i-pause-running-services-e-g-the-robot) question in the FAQ.

sebastian-brunner, 2020-09-18:
Hi @Rohith-Sewts,
you can "unsubscribe" from a topic in ROS, see https://answers.ros.org/question/239944/unsubscribing-from-ros-topic-python/

sebastian-brunner, 2020-09-18:
Concerning the ROS-node problem: The possibilities to treat ROS nodes in python are quite limited unfortunately. You can create a ros node only once in a process and this will persist until the process dies. There is no way to delete a ROS node and to restart it inside the same process. Please use this state to init and acccess the ros node in RAFCON: https://github.com/DLR-RM/RAFCON-ros-state-machines/tree/master/init_ros_node

sebastian-brunner, 2020-09-18:
@Rohith-Sewts : In which company are you using RAFCON currently?

ShuSewts, 2020-09-24:
> @Rohith-Sewts : In which company are you using RAFCON currently?

@sebastian-brunner I am  working for [https://www.sewts.de/](url) and I am designing the state machine with RAFCON for our process...I tried unregister before I posted the question. The subscribers still run in the background.