DLR-RM/RAFCON #849: Missing 'Default value' columns in editor

sillkjc, 2021-05-27:
Hi all,
I'm attempting to do the 99 bottles tutorial. I ran into a problem where the 'Default value' column for scoped variables and ports seems to be missing. Maybe this is related to my desktop environment (KDE Plasmashell 5.12.9, Ubuntu 18.04)?
Has anyone had a similar problem and found a way to make it display?

RAFCON 0.14.6 installed via Pip.

[image]

sillkjc, 2021-05-27:
Nevermind, found under separate Data Linkage tab in state editor.

closed 2021-05-27 by sillkjc

sebastian-brunner, 2022-11-18:
The state editor of @sillkjc shows the "Logical Linkage" tab. You can select the "Data Linkage" tab just below. There you can manage your data (input, output and for container states also scoped variables).

mfrigerio17, 2022-11-18:
Thanks I finally found it. Guys, the problem is that the interface shows the title "data linkage" AND "logical linkage" at the bottom of the state editor, even when the "logical linkage" tab is active... This is very confusing.

For the others with the same problem as me: you can switch mode(tab) with the small blue buttons in the top left of the state editor (e.g. the one that looks like a section of a chain)