DLR-RM/RAFCON #852: installation error

jeffRowley321, 2021-09-14:
I just wanted to install Rafcon on my machine (Ubuntu 20.04) (tried python 3.7 and 3.8 for installation)
I followed [these](https://github.com/DLR-RM/RAFCON) installation steps but I get this error when I try to run the GUI with command `rafcon` in the terminal after installation:

```
2021-09-14 14:11:52:  WARNING - rafcon.gui:  Could not import all packages required for opening the GUI: Namespace GtkSource not available for version 3.0
2021-09-14 14:11:52:    DEBUG - rafcon.core.library_manager:  Initializing Storage object ...
2021-09-14 14:11:52:    DEBUG - rafcon.core.execution.execution_status:  State machine status is set to STATE_MACHINE_EXECUTION_STATUS.STOPPED
2021-09-14 14:11:52:    DEBUG - rafcon.core.execution.execution_engine:  State machine execution engine initialized
2021-09-14 14:11:52:     INFO - rafcon.gui.utils.splash_screen:  Starting RAFCON...
2021-09-14 14:11:52:     INFO - rafcon.gui.utils.splash_screen:  Setting up logger...
2021-09-14 14:11:52:     INFO - rafcon.gui.utils.splash_screen:  Initializing plugins...
2021-09-14 14:11:52:     INFO - rafcon.gui.utils.splash_screen:  Setting up environment...
2021-09-14 14:11:52:     INFO - rafcon.gui.utils.splash_screen:  Loading configurations...
2021-09-14 14:11:52:    DEBUG - rafcon.core.config:  Configuration loaded from /home/usr/.config/rafcon/config.yaml
2021-09-14 14:11:52:    DEBUG - rafcon.core.library_manager:  Initializing LibraryManager: Loading libraries ... 
2021-09-14 14:11:52:    DEBUG - rafcon.core.library_manager:  Adding library root key 'advanced_examples' from path '/home/usr/.local/share/rafcon/examples/functionality_examples'
2021-09-14 14:11:52:    DEBUG - rafcon.core.library_manager:  Adding library root key 'generic' from path '/home/usr/.local/share/rafcon/libraries/generic'
2021-09-14 14:11:52:    DEBUG - rafcon.core.library_manager:  Adding library root key 'ros' from path '/home/usr/.local/share/rafcon/examples/libraries/ros_libraries'
2021-09-14 14:11:52:    DEBUG - rafcon.core.library_manager:  Adding library root key 'turtle_libraries' from path '/home/usr/.local/share/rafcon/examples/libraries/turtle_libraries'
2021-09-14 14:11:52:    DEBUG - rafcon.core.library_manager:  Adding library root key 'tutorials' from path '/home/usr/.local/share/rafcon/examples/tutorials'
2021-09-14 14:11:52:    DEBUG - rafcon.core.library_manager:  Initialization of LibraryManager done
2021-09-14 14:11:52:    DEBUG - rafcon.gui.config:  Configuration loaded from /home/usr/.config/rafcon/gui_config.yaml
2021-09-14 14:11:52:    DEBUG - rafcon.gui.runtime_config:  Configuration loaded from /home/usr/.config/rafcon/runtime_config.yaml
2021-09-14 14:11:52:     INFO - rafcon.gui.utils.splash_screen:  Loading GUI...
Traceback (most recent call last):
  File "/home/usr/.local/lib/python3.8/site-packages/astroid/astpeephole.py", line 17, in <module>
    _TYPES = (_ast.Str, _ast.Bytes)
AttributeError: module '_ast' has no attribute 'Str'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/usr/.local/bin/rafcon", line 8, in <module>
    sys.exit(main())
  File "/home/usr/.local/lib/python3.8/site-packages/rafcon/gui/start.py", line 370, in main
    setup_gui()
  File "/home/usr/.local/lib/python3.8/site-packages/rafcon/gui/start.py", line 212, in setup_gui
    from rafcon.gui.controllers.main_window import MainWindowController
  File "/home/usr/.local/lib/python3.8/site-packages/rafcon/gui/controllers/main_window.py", line 46, in <module>
    from rafcon.gui.controllers.states_editor import StatesEditorController
  File "/home/usr/.local/lib/python3.8/site-packages/rafcon/gui/controllers/states_editor.py", line 30, in <module>
    from rafcon.gui.controllers.state_editor.state_editor import StateEditorController
  File "/home/usr/.local/lib/python3.8/site-packages/rafcon/gui/controllers/state_editor/state_editor.py", line 33, in <module>
    from rafcon.gui.controllers.state_editor.source_editor import SourceEditorController
  File "/home/usr/.local/lib/python3.8/site-packages/rafcon/gui/controllers/state_editor/source_editor.py", line 26, in <module>
    from pylint import lint
  File "/home/usr/.local/lib/python3.8/site-packages/pylint/lint.py", line 64, in <module>
    import astroid
  File "/home/usr/.local/lib/python3.8/site-packages/astroid/__init__.py", line 67, in <module>
    from astroid.builder import parse, extract_node
  File "/home/usr/.local/lib/python3.8/site-packages/astroid/builder.py", line 26, in <module>
    from astroid import rebuilder
  File "/home/usr/.local/lib/python3.8/site-packages/astroid/rebuilder.py", line 17, in <module>
    from astroid import astpeephole
  File "/home/usr/.local/lib/python3.8/site-packages/astroid/astpeephole.py", line 19, in <module>
    _TYPES = (_ast.Str, )
AttributeError: module '_ast' has no attribute 'Str'

```

Help would be much appreciated

franzlst, 2021-09-14:
Thank you for reporting this issue. We depend on pylint 1.*, which seems to be not compatible with Python >3.6. However, this exception is raised by astroid (a dependency of pylint). Could you please test whether `pip3 install astroid --upgrade` fixes the issue?

jeffRowley321, 2021-09-14:
Thanks for your help. When I try this I get the following exception:

```
Exception:
Traceback (most recent call last):
  File "/home/usr/.local/lib/python3.8/site-packages/pip/basecommand.py", line 209, in main
    status = self.run(options, args)
  File "/home/usr/.local/lib/python3.8/site-packages/pip/commands/install.py", line 249, in run
    with self._build_session(options) as session:
  File "/home/usr/.local/lib/python3.8/site-packages/pip/basecommand.py", line 66, in _build_session
    session = PipSession(
  File "/home/usr/.local/lib/python3.8/site-packages/pip/download.py", line 320, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/home/usr/.local/lib/python3.8/site-packages/pip/download.py", line 92, in user_agent
    zip(["name", "version", "id"], platform.linux_distribution()),
AttributeError: module 'platform' has no attribute 'linux_distribution'
Traceback (most recent call last):
  File "/home/usr/.local/bin/pip3", line 8, in <module>
    sys.exit(main())
  File "/home/usr/.local/lib/python3.8/site-packages/pip/__init__.py", line 217, in main
    return command.main(cmd_args)
  File "/home/usr/.local/lib/python3.8/site-packages/pip/basecommand.py", line 242, in main
    with self._build_session(
  File "/home/usr/.local/lib/python3.8/site-packages/pip/basecommand.py", line 66, in _build_session
    session = PipSession(
  File "/home/usr/.local/lib/python3.8/site-packages/pip/download.py", line 320, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/home/usr/.local/lib/python3.8/site-packages/pip/download.py", line 92, in user_agent
    zip(["name", "version", "id"], platform.linux_distribution()),
AttributeError: module 'platform' has no attribute 'linux_distribution'
```

franzlst, 2021-09-15:
Hmm, then please try updating pip, wheels and setuptools first:

`pip install --upgrade pip setuptools wheel`

jeffRowley321, 2021-09-16:
Thanks for the information. I did the update step but when I run `rafcon ` the error from above still persists.

franzlst, 2021-09-16:
May `pip` links to `pip2` and not `pip3`.

Try 

`pip3 install --upgrade pip setuptools wheel`

or 

`python3.8 -m pip install --upgrade pip setuptools wheel`

instead.

jeffRowley321, 2021-09-16:
Ok I think we could fix the issue. The steps I did:
switched to
```
$ python --version
Python 3.7.12
```
then to 
```
$ pip --version
pip 21.2.4 from /home/usr/my_env_project/lib/python3.8/site-packages/pip (python 3.8)
```
and finally your very first hint with
`pip3 install astroid --upgrade`

Thanks a lot for your time, mate

franzlst, 2021-09-17:
I'm happy to here that it is working now!

Thank you for sharing the solution.

You output from step 2 is probably misleading, as it is showing the pip version used by python 3.8, not 3.7. I think you need to call `python3.7 -m pip --version` to get the desired result.

closed 2021-09-22 by franzlst