DLR-RM/RAFCON #860: Relative path to other python file in rafcon project

samuel-six, 2022-01-04:
Hi,

I wonder what is the best way to access other files from a python file inside a rafcon state with a relativ path. There are a few different ways to work with relative paths in python but it seems that it does not work here due to file system.

I already managed to access other files with absolute paths but this is not what I really want.

Thanks!

franzlst, 2022-01-05:
For me it would help if you could describe your use case a bit further. Are you talking of Library Sates? Does your libraries ship with additional files, which you want to access?

samuel-six, 2022-01-06:
Let's say I have an execution state with the associated .py file. From here I would like to access another python file, which is located in a folder I created myself, one or two levels further outside in the folder hierarchy. This problem seems trivial at first glance, but I've tried a few approaches in Python and only the absolute path access works at the moment.

franzlst, 2022-01-07:
The `execute` scripts has a parameter `self`, pointing to the state object. It has a member `file_system_path`, which is the absolute path of the state on the file system. From this path, you can navigate relatively, I guess:

```python
import os
def execute(self, inputs, outputs, gvm):
    self.logger.info(os.path.join(self.file_system_path, ".."))
```

closed 2022-01-07 by samuel-six