DLR-RM/RAFCON #870: Error import external library (ROS2)

Hytac, 2022-09-16:
Hi, I have a working ros2 package with all the interfaces for actions and services.
I want to call a service/action from a rafcon ExecutionState, its working fine with interfaces that are already define in ros2 such as (std_msgs/String.msg) but i can not make it work with my custom interfaces.

When i try to import the library in my python code i get the following error:

`2022-09-16 05:00:49:    DEBUG - rafcon.gui.controllers.state_editor.source_editor:  Parsing execute script...
[
    {
        "type": "error",
        "module": "file_to_get_pylinted",
        "obj": "",
        "line": 8,
        "column": 0,
        "path": "/tmp/rafcon-usr/4068/tmpaz0by0k4/file_to_get_pylinted.py",
        "symbol": "import-error",
        "message": "Unable to import 'SetRGB'",
        "message-id": "E0401"
    }
]
`

This is where my library is located:
[image]

I have added my library to rafcon like this:
[image]

My ExecutionState code looks like this:
[image]

franzlst, 2022-10-19:
I'm not so into ROS 2, but in general when you have import errors, you need to make sure your `PYTHONPATH` env var is properly set up. In your case, the  `PYTHONPATH` need to contain the parent folder of `action_server_interfaces`.

However, in your case it complains about not finding `SetRGB` class and not about missing `action_server_interfaces` module. So maybe the import path is wrong?

> I have added my library to rafcon like this:

This config is for adding RAFCON libraries, not for adding Python packages.

sebastian-brunner, 2022-11-17:
Closing due to inactiveness. Please reopen if requried.

closed 2022-11-17 by sebastian-brunner