DLR-RM/RAFCON #899: How to access exceptions on aborted outcomes?

sillkjc, 2024-12-12:
Hi all,
The RAFCON documentation states that exceptions raised within states will flow up to their parents until the 'Aborted' outcome is connected into a state.
Is there a way to get access to the original Exception object (from the child) inside the state the aborted outcome connects to? We wish see the Exception type, message, error code etc. so we can handle it appropriately with a high level exception handler state.

`For hierarchy states, the child states together with the transitions determine the outcome. As soon as during the execution of a hierarchy state a transitions leads to an outcome of the hierarchy state, the state is left on this outcome. However, if an exception or error occurs in a child state, which is not caught (by connecting the “aborted” outcome of the erroneous state), the hierarchy state is left also on the “aborted” outcome. By this, errors and exception climb up the hierarchy, until an “aborted” outcome is connected or the root state is reached (and the program stops).`

Many thanks

sebastian-brunner, 2024-12-12:
Yes you can. It's mapped to `inputs["error"]` of the abortion-handling state.

There is also an example here: https://github.com/DLR-RM/RAFCON/tree/master/source/rafcon/share/rafcon/examples/functionality_examples/hierarchy_abortion_handling

JohannesErnst, 2025-01-08:
@sillkjc Did this work for you? If so please close the issue

closed 2025-01-08 by sillkjc