DLR-RM/RAFCON #989: installation fails (relies on 'formatargspec' which was removed after python3.10)

BMaxV, 2025-01-12:
Hi everyone,

the install instructions don't work for me, I'm on Ubuntu 24.04 and that's using python3.12

The first problem was the cairo version, which I solved by trying the installation in a venv, which worked, but now I'm running into an error that ends like this:

```
...
File "/tmp/pip-install-l7u3d3ai/wrapt_6be4374b6f7f4b05b02ded7eb9a43786/src/wrapt/__init__.py", line 10, in <module>
          from .decorators import (adapter_factory, AdapterFactory, decorator,
        File "/tmp/pip-install-l7u3d3ai/wrapt_6be4374b6f7f4b05b02ded7eb9a43786/src/wrapt/decorators.py", line 34, in <module>
          from inspect import ismethod, isclass, formatargspec
      ImportError: cannot import name 'formatargspec' from 'inspect' (/usr/lib/python3.12/inspect.py). Did you mean: 'formatargvalues'?
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

And that seems to be a problem because "formatargspec" was deprecated in 3.5 and removed after python3.10

https://github.com/python/cpython/issues/89483

I would have loved to check out the project, maybe this can be updated?

if I missed something else, please let me know.

Thanks!

JohannesErnst, 2025-01-13:
Thanks for posting the issue and finding this incompatibility. Right now we are still using python 3.9 and thus didn't come across this yet. We will take a look!

For now, maybe you can just create a venv with python 3.10 or lower?

JohannesErnst, 2025-01-14:
This should be fixed in https://github.com/DLR-RM/RAFCON/commit/6f2ff844d68f61163159a3248ec903f900548a49, although I cannot thoroughly test it right now as I am lacking access to an Ubuntu System with python 3.12. 

The update will be deployed with the new minor version `2.2.0` within the next days.

JohannesErnst, 2025-01-17:
We released the new version 2.2.0: https://github.com/DLR-RM/RAFCON/releases/tag/2.2.0

Could you @BMaxV please check if it works now via `pip` with your setup?

BMaxV, 2025-01-17:
Hi,

I can confirm that the original problem was resolved, however trying to install and run it, I was also encountering this:

```
../meson.build:31:9: ERROR: Dependency 'gobject-introspection-1.0' is required but not found.
```

which was fixed by installing `libgirepository1.0-dev` because I'm on ubuntu

After that the build and install appeared to work? There were no install / build errors anymore.

Now launching:

I had to install `setuptools` with pip which was not an issue.

Then I'm running into this:

```
...
2025-01-17 20:33:33:     INFO - rafcon.gui.utils.splash_screen:  Loading GUI...
Traceback (most recent call last):
  File "/home/max/Documents/Python/rafcon/myvenv/bin/rafcon", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/max/Documents/Python/rafcon/myvenv/lib/python3.12/site-packages/rafcon/gui/start.py", line 437, in main
    setup_gui()
  File "/home/max/Documents/Python/rafcon/myvenv/lib/python3.12/site-packages/rafcon/gui/start.py", line 256, in setup_gui
    from rafcon.gui.controllers.main_window import MainWindowController
  File "/home/max/Documents/Python/rafcon/myvenv/lib/python3.12/site-packages/rafcon/gui/controllers/main_window.py", line 45, in <module>
    from rafcon.gui.controllers.states_editor import StatesEditorController
  File "/home/max/Documents/Python/rafcon/myvenv/lib/python3.12/site-packages/rafcon/gui/controllers/states_editor.py", line 30, in <module>
    from rafcon.gui.controllers.state_editor.state_editor import StateEditorController
  File "/home/max/Documents/Python/rafcon/myvenv/lib/python3.12/site-packages/rafcon/gui/controllers/state_editor/state_editor.py", line 33, in <module>
    from rafcon.gui.controllers.state_editor.source_editor import SourceEditorController
  File "/home/max/Documents/Python/rafcon/myvenv/lib/python3.12/site-packages/rafcon/gui/controllers/state_editor/source_editor.py", line 41, in <module>
    from rafcon.gui.views.state_editor.source_editor import SourceEditorView
  File "/home/max/Documents/Python/rafcon/myvenv/lib/python3.12/site-packages/rafcon/gui/views/state_editor/source_editor.py", line 16, in <module>
    from gi.repository import GtkSource
  File "/home/max/Documents/Python/rafcon/myvenv/lib/python3.12/site-packages/gi/importer.py", line 133, in create_module
    raise ImportError('cannot import name %s, '
ImportError: cannot import name GtkSource, introspection typelib not found
```

Is this enough error log, or do you need more?

Thanks for the quick fix! A week is amazing!

JohannesErnst, 2025-01-18:
Hi, thanks for the update. I am sorry that the changes didn't fully fix the installation problem. Unfortunately, we are currently lacking the resources to test the installation with the mentioned system specifications.

I hope that we will, in the future, be able to provide a stable release for ubuntu and latest python versions. In the meantime, if you happen to figure out a way to install and launch it, hopefully you can post some update here. Maybe the solution with a virtual environment for e.g. python 3.9 works...

BMaxV, 2025-01-18:
Alright, thank you very much for your effort, I think I will just close it then. The original problem was fixed after all.

closed 2025-01-18 by BMaxV

BMaxV, 2025-11-02:
Hey,

I followed the instructions, including the `install the rafcon[gtk]`, but when I try to start it, I get this warning:

```
WARNING - rafcon.gui:  Could not import all packages required for opening the GUI: Namespace GtkSource not available
```

and then it immediately exits again. Cleanly though. Are there additional debug options I can use to figure out why it's missing?

JohannesErnst, 2025-11-10:
Newest version 2.3.1 and updated install instructions are released.