DLR-RM/RAFCON #993: Zooming in too far causes display to become blank

brta-jc, 2025-02-11:
Hi all, 
Thank you for any help in advance.

My team is having an issue with the GUI editor when we zoom. At one zoom level we have this:
[image]

Once we zoom in one more level / tick on the scroll wheel, the state machine representation / window goes blank. The rest of RAFCON is fine, like so:

[image]

`
2025-02-11 16:55:34:     INFO - rafcon.gui.controllers.graphical_editor_gaphas:  Opening the state machine caused some meta data to be generated, which will be stored  when the state machine is being saved.
2025-02-11 16:55:34:  VERBOSE - rafcon.gui.controllers.graphical_editor_gaphas:  Time spent in setup canvas 0.04892301559448242 state machine 1
2025-02-11 16:55:35:     INFO - rafcon.gui.controllers.main_window:  Ready

(ros2:5727): Pango-WARNING **: 16:55:42.960: failed to create cairo scaled font, expect ugly output. the offending font is 'Font Awesome 5 Free Heavy 59622.3349609375'

(ros2:5727): Pango-WARNING **: 16:55:42.960: font_face status is: error occurred in libfreetype

(ros2:5727): Pango-WARNING **: 16:55:42.960: scaled_font status is: error occurred in libfreetype
Traceback (most recent call last):
  File "/home/ros/.local/lib/python3.10/site-packages/gaphas/decorators.py", line 137, in async_wrapper
    func(*args, **kwargs)
  File "/home/ros/.local/lib/python3.10/site-packages/gaphas/view.py", line 796, in update_back_buffer
    self.painter.paint(Context(cairo=cr, items=items))
  File "/home/ros/.local/lib/python3.10/site-packages/gaphas/painter.py", line 74, in paint
    painter.paint(context)
  File "/home/ros/.local/lib/python3.10/site-packages/gaphas/painter.py", line 143, in paint
    self.draw_items(context.items, cairo)
  File "/home/ros/.local/lib/python3.10/site-packages/gaphas/painter.py", line 120, in draw_items
    self.draw_item(item, cairo)
  File "/home/ros/.local/lib/python3.10/site-packages/gaphas/painter.py", line 113, in draw_item
    cairo.restore()
cairo.Error: error occurred in libfreetype

(ros2:5727): Gtk-WARNING **: 16:55:42.961: drawing failure for widget 'GtkLabel': error occurred in libfreetype
(ros2:5727): Gtk-WARNING **: 16:55:42.961: drawing failure for widget 'GtkButton': error occurred in libfreetype
(ros2:5727): Gtk-WARNING **: 16:55:42.961: drawing failure for widget 'GtkBox': error occurred in libfreetype
(ros2:5727): Gtk-WARNING **: 16:55:42.961: drawing failure for widget 'GtkEventBox': error occurred in libfreetype
(ros2:5727): Gtk-WARNING **: 16:55:42.961: drawing failure for widget 'GtkNotebook': error occurred in libfreetype
`

When you zoom back out, the display recovers and is normal again. RAFCON continues to run fine, with no glitches other than the error log and visual element disappearing. 
It looks like RAFCON is having trouble scaling the fonts? As above, the scale was 59622.3349609375, at a certain threshold it fails to scale and errors.

One thing to note:
- We're running on Ubuntu22 via WSL2, and most everything works great. We had some GTK issues causing the RAFCON GUI to crash using the default WSL2 window manager Wayland. The crash was a little bit similar, in that elements of the display were lost. However, in this case the entire program would eventually crash. Forcing WSL to use X11 (with VNC forwarding to windows) solved them, via export GDK_BACKEND=x11

Thanks again!

brta-hj, 2025-02-12:
Hi @JohannesErnst,

Thanks for the suggestions above. I have tried the font replacement (by pointing the `ICON_FONT_AWESOME` parameter in `rafcon/gui/utils/constants.py`.

This seems to have resolved the error about `Font Awesome 5 Free Heavy 59622.3349609375`, but now I get the error for the following font (which does not seem to be part of RAFCON fonts):
```
(ros2:20211): Pango-WARNING **: 11:30:26.683: failed to create cairo scaled font, expect ugly output. the offending font is 'DejaVu Sans 59622.3349609375'
(ros2:20211): Pango-WARNING **: 11:30:26.683: font_face status is: error occurred in libfreetype

(ros2:20211): Pango-WARNING **: 11:30:26.683: scaled_font status is: error occurred in libfreetype

(ros2:20211): Pango-WARNING **: 11:30:26.871: failed to create cairo scaled font, expect ugly output. the offending font is 'DejaVu Sans 73607.80859375'

(ros2:20211): Pango-WARNING **: 11:30:26.871: font_face status is: error occurred in libfreetype

(ros2:20211): Pango-WARNING **: 11:30:26.871: scaled_font status is: error occurred in libfreetype
```

As for the visual issue in the GUI, it remains. The state machine window still goes blank at that highly zoomed in level, and the terminal logs a lot of `error occurred in libfreetype` messages.

(I am also running in X11 mode on WSL2 as per @brta-jc message above)

brta-hj, 2025-02-12:
I experience the same issue in the "99_bugs" included tutorial state machine, if that helps to reproduce on your end. Just zooming in a lot on any one of the internal blocks causes the errors to appear, this time the font called out is `Source Sans Pro`.

Once the initial error occurs, I get the following Gtk-WARNING errors indefinitely when changing the zoom or moving the view around:

```
(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkLabel': error occurred in libfreetype

(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkBox': error occurred in libfreetype

(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkEventBox': error occurred in libfreetype

(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkNotebook': error occurred in libfreetype

(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkBox': error occurred in libfreetype

(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkPaned': error occurred in libfreetype

(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkBox': error occurred in libfreetype

(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkPaned': error occurred in libfreetype

(rafcon:17455): Gtk-WARNING **: 12:47:08.396: drawing failure for widget 'GtkPaned': error occurred in libfreetype

```

JohannesErnst, 2025-02-12:
Hi @brta-hj,

okay, thanks for testing it. From that I conclude that it has nothing to do with the font directly. Although `DejaVu Sans` and `Source Sans Pro` don't appear in the RAFCON code directly, maybe they are still being used but selected via a different string. 

I cannot reproduce the same behavior that you experience, even with the `99_bugs` state machine. I don't think it is connected to the state machine specifically but rather generally to how data ports, connection lines and state names are scaled on high zooming levels. 
Therefore, I at least investigated the `Pango-CRITICAL` error that gets thrown in my setup. Even though it doesn't lead to problems on my end, it might be the cause for the issue on your end when put in combination with the additional environment overhead you are using. 
I spend some hours debugging the code and found that the `layout = PangoCairo.create_layout(cairo_context)` seems to be the culprit. I detected problematic usages in the following three files:
- https://github.com/DLR-RM/RAFCON/blob/99e0daba66d7748346fb0cfa18a5f65f2ab490bf/source/rafcon/gui/mygaphas/items/state.py#L502
https://github.com/DLR-RM/RAFCON/blob/99e0daba66d7748346fb0cfa18a5f65f2ab490bf/source/rafcon/gui/mygaphas/items/state.py#L525-L530
- https://github.com/DLR-RM/RAFCON/blob/99e0daba66d7748346fb0cfa18a5f65f2ab490bf/source/rafcon/gui/mygaphas/items/line.py#L263-L267
https://github.com/DLR-RM/RAFCON/blob/99e0daba66d7748346fb0cfa18a5f65f2ab490bf/source/rafcon/gui/mygaphas/items/line.py#L285
- https://github.com/DLR-RM/RAFCON/blob/99e0daba66d7748346fb0cfa18a5f65f2ab490bf/source/rafcon/gui/mygaphas/utils/gap_draw_helper.py#L118-L127
https://github.com/DLR-RM/RAFCON/blob/99e0daba66d7748346fb0cfa18a5f65f2ab490bf/source/rafcon/gui/mygaphas/utils/gap_draw_helper.py#L192

Right now I cannot spend more time on this, feel free to investigate yourself. The lines above can serve as a reference on where to continue debugging. My suspicion is that the `cairo_context` that is passed in each case is somehow parameterized in a bad way causing problems when trying to e.g. `layout.get_extents()` or `PangoCairo.show_layout(cairo_context, layout)`

brta-hj, 2025-02-12:
Hi @JohannesErnst,

Great, thanks for investigating it a bit further for us. I will have a look at the sections you mention above and try to debug further.

If possible, can you please share what your environment setup is like?

brta-hj, 2025-02-13:
Just a small update: I've replaced all occurrences of `font-family: 'Source Sans Pro'` in the various `css` files in `rafcon/share/themes` folder with `font-family: Monospace` and that manages to get rid of the flood of `Gtk-WARNING **` errors that were previously occuring after the initial scaled font error. Not really addressing the source of the problem here, but it does seem to have made the app more stable for me now.

As a sanity check, I have tried RAFCON in Ubuntu 20.04 on a VM I had and get only the `Pango-CRITICAL` error that you experience in your setup.

JohannesErnst, 2025-02-13:
Hi @brta-hj,

good to know that you experience the same issue in a "normal" Linux environment without the additional Wayland layer. Currently, I am running on a _openSUSE Leap 15.5_ but without any virtual machine or WSL. I expect to test more on Ubuntu also in the future. 

Interesting to know that the `Source Sans Pro` seems to be causing the `Gtk-WARNING` on your system. The error source is probably in the same direction.

As the original "blank display issue" does indeed seem to be connected to WSL2, maybe it is worth trying some different fonts or tweaking the env variables for this...

gilopez01, 2025-05-29:
Thanks for the quick reply.

In my case that variable never drops to 0 (so the guard you mentioned is never hit) and the crash happens in a high zoom level, but not too high, my state machine has about three depth levels, and using this state machine I reach that zoom level regularly.

What I’ve tried so far:

* Capped `MAX_ZOOM` inside `ExtendedGtkView.zoom()`.
* Wrapped the calls to `draw_name` / `draw_symbol` in `try/except: pass`.
* Added a similar `try/except: pass` around every `c.restore()`.

The real problem is that once I hit the failing zoom the entire UI goes blank; when I zoom back out the canvas repaints, but all other widgets stay white. Is there any hot-fix that would isolate the center canvas from the rest of the UI so that only the canvas goes blank (and can be recovered by zooming out) while the rest of the interface remains usable?

JohannesErnst, 2025-06-06:
So I investigated a little bit further and found some more maybe interesting points:
- To confirm again, the problems seem to solely be connected to **scaling the font** of texts in the graphical editor. This includes, e.g. port names, data connections, state names etc. When I remove all the functions to draw texts, also the `pango-CRITICAL` disappears. This, of course is not a solution, however.
- Reading carefully through this issue, I noticed that both for @gilopez01 and @brta-jc the problem seems to occur with Ubuntu 22. On Ubuntu 20, I think @brta-hj confirmed that they also just got the `pango-CRITICAL` without any direct problems in the GUI. Maybe the Ubuntu version is connected to the problem.
- I tracked one of the `pango-CRITICAL` errors to this line: https://github.com/DLR-RM/RAFCON/blob/99e0daba66d7748346fb0cfa18a5f65f2ab490bf/source/rafcon/gui/mygaphas/items/line.py#L265 The layout is created in the function call above here: https://github.com/DLR-RM/RAFCON/blob/99e0daba66d7748346fb0cfa18a5f65f2ab490bf/source/rafcon/gui/mygaphas/utils/gap_draw_helper.py#L287-L297 Interestingly, the function `get_text_layout()` does not raise any warnings or errors but only later when `get_extents()` is called the `pango-CRITICAL` is thrown.
It would be interesting if in your setup, @gilopez01, you even ever get to these `pango-CRITICAL` errors or if it just already breaks before? 
``` (start.py:4097): Pango-CRITICAL **: 16:08:40.177: pango_font_description_set_size: assertion 'size >= 0' failed ```

JohannesErnst, 2025-07-31:
Finally, we got a fix working on our end when using Ubuntu systems.

@gilopez01 @brta-hj @brta-jc can you please check if the fix in 
https://github.com/DLR-RM/RAFCON/tree/fix/extreme-zoom-blank-display 
solves the problem on your end?

Happy to receive your feedback! If everything works fine we will create a new release.

commit 7aa606d7 (2025-07-15): fix(gaphas): prevent blank display and crashes at extreme zoom levels

Prevents RAFCON GUI from going blank when zooming to extreme levels by:
- Clamping font sizes (0.1-1000) to stay within FreeType limits
- Adding zoom boundaries (0.0001-100000) to prevent extreme values
- Implementing safe Cairo context restoration
- Forcing cache updates for small states at high zoom
- Filtering Pango warnings from terminal (process-specific via os.dup2)

Technical: FreeType has ~16384px glyph limit. Added text_rendering.py module
for centralized safety checks. Image cache clears at zoom boundaries for
pro [... cut]
commit 51303609 (2025-07-30): fix(helpers): remove redundant font size clamping and refactor Pango warning filter

- Remove font size clamping in label.py that caused infinite loop and frozen screen on openSUSE
- The clamping fix was redundant as newer methods handle font size issues for icons/labels
- Extract Pango warning filter from start.py into dedicated warning_filters.py utility module
- Add Ubuntu-specific OS detection to only apply Pango filter on Ubuntu systems
- Improves code organization and prevents platform-specific issues

Closes #993